prusa3d/Prusa-Firmware #3242: [BUG] Can't connect Octoprint with SD card attached

nirvdrum, 2021-07-18:
**Printer type** - MK3S
**Printer firmware version**-  3.10.1-RC1

**MMU Upgrade** - MMU2S
**MMU upgrade firmware version**- 1.0.6

**SD card or USB/Octoprint**

I generally print from the SD card.

**Describe the bug**

I am unable to connect Octoprint to the printer with my SD card attached. The SD card has 97 files, comprising 463 MB.

With the SD card inserted, attempts at connecting Octoprint timeout. The connect attempt restarts the printer, which then rescans the SD card. The SD card scan must take too long, so Octoprint times out. If I remove the SD card, I can connect without issues.
  
**To Reproduce**

Fill an SD card up close to the MK3S file limit. Then try to connect Octoprint.
  
**Expected behavior**

I expected the Octoprint connection to succeed. I'm pretty sure I was able to do so with previous versions of the firmware.
  
**G-code**
N/A

JohnnyDeer, 2021-07-20:
Hi @nirvdrum, I´m unable to reproduce this issue, even if i reached max files on SD all works correct. Did you tried another SD?

nirvdrum, 2021-07-20:
Hmm... okay. I'm using the SD card that shipped with my MK3S, which is a 16 GB UHS-1 goodram card. It's been functional, so I haven't thought to switch it out. I thought I'd file the issue because the problem happens with the card sent by Prusa and the contents are within the file limit of the printer firmware. The inability to connect due to timeout can be annoying to diagnose and looks like it may be a communication issue with the printer. I spent some time trying to figure out why I couldn't connect before noticing the timeout only happened when the printer was reading the SD card file list.

Given the difference in card speeds and capacities, you may need to adjust the card contents to reproduce. But, it should be possible to artificially reproduce by adding a `sleep` call to the SD card list procedure. I can file an Octoprint issue to increase the default connection timeout, if you think that's the better solution. I think it'd be good for the stock configuration to work well and I'm hoping to save other users similar connection frustrations.

DRracer, 2021-07-29:
@nirvdrum could you please share the image of your SD card so that we can find the potential bug you are experiencing?

JohnnyDeer, 2021-08-19:
This issue will be closed soon if autor will not respond.

nirvdrum, 2021-08-20:
I'm sorry, I missed the previous message on this. What do you need for an image? I can create a ZIP file or tarball with all the contents easily enough. But, if you'd rather I use `dd` or something to a real dump, I can do that as well.

leptun, 2021-08-20:
You can use dd and then compress it since most of the image should be empty anyway.

DRracer, 2021-08-23:
@nirvdrum please do the `dd`, we need the exact bit copy of the image of your SD card's filesystem, thank you

closed 2023-11-09 by github-actions[bot]