prusa3d/Prusa-Firmware #3243: [BUG] switching back to English not possible after selecting secondary language

wschadow, 2021-07-18:
Please, before you create a new bug report, please make sure you searched in open and closed issues and couldn't find anything that matches.

**Printer type** -MK3S
**Printer firmware version**- 3.10.1-RC1

**Describe the bug**
  After selecting a secondary language it#s not possible to switch back to English. It's only possible via Factory Reset
  
**To Reproduce**
  Select any of the secondary languages. You cannot change back to English. Only when doing this via Factory Reset.

DRracer, 2021-07-19:
@wschadow this is weird, this operation works normally as it should, I just checked MK3 and MK3S

wschadow, 2021-07-21:
Thanks for tetsting. I tried on some other machines and cannot reproduce as well. Very weird, sorry for bothering.

closed 2021-07-21 by wschadow

DRracer, 2021-07-22:
Thank you for your confirmation :+1: