prusa3d/Prusa-Firmware #3244: [BUG] Restart after SD removal

KOBALT3D, 2021-07-19:
**Printer type** - [ MK3S ]
**Printer firmware version**-  [ 3.10.0 ]

**SD card or USB/Octoprint**
  No Octoprint 

**Describe the bug**
  After removing SD card from slot and selecting _No SD card_ from Main Menu printer restarts. Before restart the menu item _Main_ is displayed and after secont or two a backslash is diplayed on th second line
  
**To Reproduce**
  1. Restart or turn on the printer without SD card inserted
  2. Insert SD card with files on it (I've not tested this with empty SD card) 
  3. After file list is displayed select _Back.._ to return to _Info Screen_
  4. Remove SD card
  5. Go to Main Menu and select  _No SD card_
  
**Expected behavior**
  No file list displayed if SD card is not present.
  
**Video**
  https://drive.google.com/file/d/1qfcaugxblDfEzYGYLuM48lr42mmoZO-y/view?usp=sharing

gudnimg, 2021-07-19:
I am unable to reproduce this. I have MK3S+ though not sure if the difference is large there.

gudnimg, 2021-07-19:
@KOBALT3D It is strange in the video that it prints a backslash before the restarts. `"/"` maybe that's a hint? 🤔 I've never seen a backslash on my end (at least I don't think so).

DRracer, 2021-07-20:
@KOBALT3D please try the latest FW 3.10.1-RC1 and report back if your issue persist.
Btw. we are also unable to reproduce your issue on multiple machines.

KOBALT3D, 2021-07-20:
SOLVED - Factory Reset - Erase All

closed 2021-07-20 by KOBALT3D