prusa3d/Prusa-Firmware #3246: [QUESTION] Non-continous fast move after MBL

KOBALT3D, 2021-07-20:
Hello.

**Does anyone know why after following commands entered in OctoPrint/Repetier-Server Console the move of the head on X/Y axes is not continous?** The head moves approx. 50mm and after short pause (ms) it moves to the end position.

```
G90
G80
G1 X250 F6000
```

leptun, 2021-07-20:
@KOBALT3D When MBL is enabled, the gcode lines are split in multiple smaller segments of 30mm maximum length. The target position of each segment is adjusted by the MBL mesh. The reason the stutter occurs is because of how the planner works. It queues the segments one by one. The problem is that usually new segments depend on old segments and more importantly also the other way around. Because the stepper IRQ might have consumed the first segment by the time the second segment is queued, the first movement will use the most basic motion profile.
So the stuttering you see is basically a 30mm movement which stops completely at the end and then another longer movement to the desired target position.

@wavexx I know you looked into this some time ago. Did you find any workaround to this issue? I've noticed Klipper does something to combat this. When starting from a dry buffer, it waits a bit until new movements no longer affect the oldest move (or a timeout for when you don't have many moves).  This way they can recalculate the acceleration, start and stop positions correctly even for the first segment at the cost of a tiny delay when starting from idle.

wavexx, 2021-08-09:
There are a few places we need to tweak to combat this. There's a time during the planning of the second segment where the planner is still allowed to update the *current moving* segment, but it looks like it's not working as it should.

 Marlin2 seemed to have ditched this solution, and introduced a customizable delay for "time to first move". During this time, when starting from an empty queue, we give the planner X extra microseconds before the start the steppers. Since the motors should be moving from a full-stop anyway, this is actually simple and effective.

I remember I tried this, but it still wasn't working as expected. Probably the "wait during idle" part of the stepper code can still start the motors before we have time to set the new timer.

Needs some careful testing, but I agree this needs to be fixed. This is *very* apparent on some models if there's a travel right after a layer change (thus an empty queue)

wavexx, 2021-08-09:
I think this should be renamed to BUG, btw.

JohnnyDeer, 2021-08-19:
This issue will be closed soon if autor will not respond.

tg73, 2021-08-23:
@JohnnyDeer This issue does not appear to be waiting for author response, perhaps you could discuss with @wavexx, who appears to consider this a bug that needs to be fixed.

closed 2023-11-13 by github-actions[bot]