prusa3d/Prusa-Firmware #3249: [QUESTION] MMU2S, is the 'A' command ever used?

gudnimg, 2021-07-25:
I noticed in the Prusa Firmware that the printer can send an 'A' command. But when I look at the parsing code on the MMU side I don't see it anywhere. So I'm wondering if someone more knowledgeable about this could confirm or point where it is used?

If it is not used this code in the firmware is questionable:

```cpp
case S::WaitCmd: //response to mmu commands
        if (mmu_idl_sens)
        {
            if (READ(IR_SENSOR_PIN) == 0 && mmu_loading_flag)
            {
                DEBUG_PRINTF_P(PSTR("MMU <= 'A'\n"));
                mmu_puts_P(PSTR("A\n")); //send 'abort' request
                mmu_idl_sens = 0;
                //printf_P(PSTR("MMU IDLER_SENSOR = 0 - ABORT\n"));
            }
            //else
                //printf_P(PSTR("MMU IDLER_SENSOR = 1 - WAIT\n"));
        }
```

```cpp
case S::GetFinda: //response to command P0
        if (mmu_idl_sens)
        {
            if (READ(IR_SENSOR_PIN) == 0 && mmu_loading_flag)
            {
#ifdef MMU_DEBUG
                printf_P(PSTR("MMU <= 'A'\n"));
#endif //MMU_DEBUG  
                mmu_puts_P(PSTR("A\n")); //send 'abort' request
                mmu_idl_sens = 0;
                //printf_P(PSTR("MMU IDLER_SENSOR = 0 - ABORT\n"));
            }
            //else
                //printf_P(PSTR("MMU IDLER_SENSOR = 1 - WAIT\n"));
        }
```

DRracer, 2021-07-29:
it should be sent by the printer when the filament reaches the fsensor

gudnimg, 2021-07-29:
> it should be sent by the printer when the filament reaches the fsensor

When I look at the MMU firmware I don’t see it expecting an `‘A’` command anywhere. 

The Printer sends it but I’m worried the MMU doesn’t actually do anything in response.

DRracer, 2021-08-09:
It surely does, but it is burried deep in the spaghetti code base:
`void motion_feed_to_bondtech()...`

gudnimg, 2021-08-09:
Wow, that is well hidden 😄

closed 2021-08-10 by gudnimg