prusa3d/Prusa-Firmware #3259: [FEATURE REQUEST] Lightning "Alarm" for finished job

nicoportorico, 2021-08-04:
Hi everybody , in Order to keep tracking the job of many printers by looking at a "panoramic" camera , and spare some money (without buying raspberryes/arduino boards and such) , wouldn t it be nice to have a simple LED (right on the uppest side of the frame) that turns on once the print is done ? 

I think that s a really cheap but usefull feature to have .

DRracer, 2021-08-09:
@nicoportorico even though your request will probably not make it into production, you can still make a DYI solution. AFAIK there is a G-code for switching a pin on the CPU - you can connect an LED to it and turn it on when needed in your G-code file.

wavexx, 2021-08-09:
One odd alternative would be to flash the LCD backlight for printers that have it?

JohnnyDeer, 2021-08-19:
I´m closing this issue because there are no responses.

closed 2021-08-19 by JohnnyDeer