prusa3d/Prusa-Firmware #3261: Mesh bed levelling left-shifted

WereCatf, 2021-08-09:
Printer type: MK3S+
Printer firmware version: 3.10.0

**Describe the bug**
Mesh bed levelling is shifted to the left, ie. it seems to start at basically 0 offset on the left. On e.g. my old MK2S, mesh bed levelling is centered on the bed; the left side is probed the same distance from the edge as it is on the right side. On my MK3S+, there's VERY little distance from the left edge and on the right, it doesn't get anywhere  near the edge.

This causes e.g. the issue that firmware thinks the left edge is higher than it actually is and the right side is lower than it actually is. With bed level correction I can adjust  this to a degree, but it's still a little wonky. Besides which, one shouldn't have to use bed level correction in the first place to fix wrong mesh bed levelling.

**Expected behavior**
Mesh bed levelling is supposed to be centered on the bed.

**Video**
https://www.dropbox.com/s/jao5zeozpwshwa1/meshbedlevelling.mp4?dl=0

wavexx, 2021-08-09:
The 7x7 mesh bed leveling points should be equally distanced from the edge. It's true the extruder looks skewed to the left, but if you notice the distance of the PINDA is ~2.5cm from the left and right edge while probing also in the video.

Did you notice if you have a level shift also from bottom-to-top of the bed?

WereCatf, 2021-08-09:
@wavexx Now I feel like an idiot: I blindly stared at the nozzle instead of the PINDA and yes, the PINDA does travel right where one would expect it to. I'm just gonna close this issue and slinker away, before I make things worse.

As for the bed-issue, I'll have to dig more into then. Printer says there I only have 0.04 degrees of skew, all my tools say the bed is very level and so on, but I have to keep messing with the bed level correction and use the maximum number of Z-probes possible. With the defaults, it's just unuseable, apparently getting slightly different results every run...

closed 2021-08-09 by WereCatf

wavexx, 2021-08-09:
I think the issue in this case is that the PINDA might be getting warmed up while probing, and the built-in correction table is not 100% correct. This is why I was asking if you noticed a slight shift bottom-to-top, since the top region of the bed is also the place where the PINDA has been heated more.

Do this as a first step: in the startup gcode, home XYZ (``G28 W``), move the pinda somewhere closer to the center and nearby the plate (``G1 X100 Y100Y Z10``), sleep ~30s (``G4 S30``), then continue with the actual mesh bed leveling (``G80`` and following).

Does that make it more consistent? If so, calibrating the PINDA using G76 could fix this.

WereCatf, 2021-08-10:
@wavexx I thought the SuperPINDA doesn't use a thermistor in the first place anymore and as such there is no correction table for it, either.

wavexx, 2021-08-10:
Oh! Ok, I didn't read this was a MK3S+, sorry. I was still thinking the PINDAv2. You're right of course, in theory the SuperPINDA should be bang-on all the time.

But then I wonder again what sort of issue you're looking at. If you completely reset the manual correction, what happens if you print one of those https://www.prusaprinters.org/prints/45752-3x3-bed-level-calibration?

WereCatf, 2021-08-10:
@wavexx That's what I have been using, actually. But nevermind that: I do very much appreciate you attempting to help, but I suppose printbed-issues would better taken to the Prusa forums instead, if I can't solve things on my own.

Thank you for the attention and sorry for disturbing things here!