prusa3d/Prusa-Firmware #3262: [BUG] After inserting SD-card, the file list is sometimes automatically dismissed

wavexx, 2021-08-09:
**Printer type** - MK3+
**Printer firmware version**-  3.10.1-RC1

**MMU Upgrade** - No
**MMU upgrade firmware version** - N/A

**SD card or USB/Octoprint**
SD card

**Describe the bug**
When inserting an SD card, the file list is displayed for a fraction of a second, but then it returns immediately to the status screen before you have time to do anything.

This is a new issue that got introduced recently (3.10.1-RC1 for sure, but could be 3.10.0). Can't determine exactly, since it's somewhat rare and doesn't reproduce consistently at all.

IMHO this happens due to an imperfect SD contact while being sled-in, *probably*. But we detect this late, and dismiss the menu. 
Or maybe we simply forgot to reset the "return to status" timer after sorting?

I never worked too closely to the SD code, but maybe there's a flag we should reset? @leptun 

Because it looks like the file list is perfect so *clearly* we could read it just fine. If I return to the SD menu manually, everything is working - so SD hasn't been removed either. I'm certain the contacts are clean, make good contact as well, and do not disconnect even with vibrations.

**To Reproduce**
Common way to reproduce (reproduces 1 time in ~20 tries):

* Remove SD card
* Reset printer
*  Wait ~10 seconds
* Insert SD card, not too fast
* File sorting starts, then shows file list
* Less than a second later, the menu automatically returns to the status screen
  
**Expected behavior**
After inserting SD card, we shouldn't go back to the status screen immediately.
  
**G-code**
N/A

**Video**
Couldn't record it (yet)

wavexx, 2021-08-09:
Also pinging @gudnimg - did you notice this happening to you?

gudnimg, 2021-08-09:
> Also pinging @gudnimg - did you notice this happening to you?

No I haven’t noticed it before. But I will try to reproduce it later today after work.

just for clarification:
> Reset printer

Do you mean to turn the printer on and off. Or factory reset? Or are you referencing the reset button on the PCB?

Panayiotis-git, 2021-08-09:
It is happening for me always at the first SD card insertion, after a reset or power on.

gudnimg, 2021-08-09:
I can't seem to be able to reproduce this. Though I would expect the issue to be somehow in this function `menu_lcd_lcdupdate_func()`

gudnimg, 2021-08-09:
I am able to reproduce this without a reset.

1. Remove the SD card
2. Wait a bit 20-40 seconds
3. Insert card

Even if I insert the card as fast as I can, I can still reproduce it.

gudnimg, 2021-08-09:
I have found that this function is causing the issue: `other_menu_expired()`

and that triggers this line: https://github.com/prusa3d/Prusa-Firmware/blob/3bf5d937a85e2afdeacdd45d4f19adfaff69c14e/Firmware/ultralcd.cpp#L8984

wavexx, 2021-08-09:
I don't think it's a forced_menu_expired(), so it might be indeed other_menu_expired(). In this case, we probably need to reset the timer just before entering the SD menu. I expected this to be automatic, but maybe not.

gudnimg, 2021-08-09:
All of a sudden I can't reproduce it the same way I did before. This smells funny 🤔 hmm

wavexx, 2021-08-13:
It also does happen if you reset with the card *already* inserted, wait some random time, then remove and re-insert the card. I just had one case of this happening.

Panayiotis-git, 2021-08-16:
A couple of weeks ago, this was happening all the time. Today, I tried to reproduce it and it didn't happen once...

leptun, 2021-08-20:
Not to worry. After a few weeks of not using the MK3 firmware at all, the first SD insertion after a preheat (was removed at bootup) yielded this bug.

gudnimg, 2021-08-21:
I submitted one PR for this issue just now, it's just a proposal and may not fix it since we don't know how to reproduce this 100% of the time. But I suppose this is also what you had in mind @wavexx by restarting the timer?

I suspect I created another bug by removing the call from presort() 😅

PR #3305 (2021-11-19): Fix issue when the SD card is inserted and the file menu may sometimes close immediately due to a timeout 
Make sure to call lcd_timeoutToStatus.start() when the SD card is inserted
into the printer after it was removed beforehand.

Fixes  #3262

(cherry picked from commit 6795843f153aa064f087f80990419a946cdfe857)

Previous PR: https://github.com/prusa3d/Prusa-Firmware/pull/3269
commit be79e379 (2021-11-19): Fixes  #3262

Make sure to call lcd_timeoutToStatus.start() when the SD card is inserted
into the printer after it was removed beforehand.

(cherry picked from commit 6795843f153aa064f087f80990419a946cdfe857)