prusa3d/Prusa-Firmware #3263: [ENHANCEMENT]<Ability to change what height the filament sensor is at>

firedemontwinkie, 2021-08-10:
**Printer type** - [e.g. MK3S, MK3, MK2.5S, MK2.5, MK2S, MK2]
**MMU Upgrade** - [e.g. MMU2S, MMU2, MMU1]

**Is your enhancement related to a problem? Please describe.**
  the filament sensing meninism on the MK3S+ with the MMU2S sucks so I made a custom part that allowed the connector cable to connect to the MK3S+ extruder and use the old (and better) filament sensing system(the one with magnets) the problem is that the point that which it senses is higher up so it takes a little push on the filament to get it to work when it loading or changing filaments. I ask Pursa tech support and they say there is no way I can fix it on the slicer or setting in the printer. 
  
**Describe the solution you'd like**
  maybe in the settings you can set the time the MMU2S waits to stop sending the filament after it senses the filament.

if it not a good idea can someone at least point me to the file that deals with filament loading.
my email is firedemontwinkie@gmail.com

tg73, 2021-08-23:
@firedemontwinkie While not a direct answer to your post, might I suggest that you look at [this](https://www.prusaprinters.org/prints/66215-tzb-mk3smk25smmu2s-extruder-mmu2s-no-tower-v3) design. It's very close to the stock extruder, with modification only to better implement filament sensing. I use this design as it's the only way I can get reliable handling of hard (eg, pla) and soft (eg, flex) filaments with the same adjustment of the sensor. With the stock design, it's basically impossible to adjust the chimney and bondtech tension to work reliably for both hard and soft filaments. What exactly what "sucks" about the stock design in your experience?

firedemontwinkie, 2021-08-23:
Thank you so much for the email. I thought my post was all but forgotten.

firedemontwinkie, 2021-08-23:
the thing that sucks with the MMU2s filament sensing system on the
extruder. no matter what I did with the screw that applied tension the
sensor would not sense and the MMU2s would not stop loading. it been a
headache.

firedemontwinkie, 2021-08-23:
here is the part I made
https://www.prusaprinters.org/prints/71199-modified-fs-cover-mmu2s

closed 2023-11-13 by github-actions[bot]