prusa3d/Prusa-Firmware #3264: [BUG] Inserting/Removing the SD card while sorting can crash/get-stuck

wavexx, 2021-08-13:
**Printer type** - MK3+
**Printer firmware version**-  3.10.1-RC1
**MMU Upgrade** - n/a

**SD card or USB/Octoprint**
SD

**Describe the bug**
The "Sorting files" progress bar can get stuck if you remove and re-insert the SD card
  
**To Reproduce**
- Remove SD card
- Reset printer
- Insert SD card
- Remove SD card while "Sorting files"
- Quickly re-insert the SD card

At this point, sorting might either 1) get stuck forever 2) continue happily but then show "card removed" 3) stop sorting, and restart correctly 4) get stuck and crash 5) continue and show a broken SD menu (and crash shortly afterwards)

All options available :)
  
**Expected behavior**
The FW shouldn't get stuck.

I think the issue in this case is that the menu_lcl_lcdupdate_func() re-triggers the sorting code while the old didn't finish yet.

**Video**

https://user-images.githubusercontent.com/1017726/129354056-ce5c8a35-f54f-4029-bfa8-d260966bd250.mp4

gudnimg, 2023-06-22:
@wavexx wasn't this issue fixed in 3.10.1? Can this issue be closed?

wavexx, 2023-06-22:
I know we had some fixes, but I didn't retest

closed 2023-11-13 by github-actions[bot]