prusa3d/Prusa-Firmware #3265: [BUG] 3.10.1-rc1 - Long-press during print doesn't respond

dalgibbard, 2021-08-15:
**Printer type** - MK3S+ with MMU2S
**Printer firmware version**-  3.10.1-rc1-4587

**MMU Upgrade** - MMU2S
**MMU upgrade firmware version** - 1.0.6-372

**SD card or USB/Octoprint**
Octoprint 1.6.1

**Describe the bug**
During an active print, long-press to adjust Z-height is not working. Menu is displayed instead, regardless of how long the button is held.
  
**To Reproduce**
Start print. Long press button.
  
**Expected behavior**
Long press should show the Z-height tune menu.
  
**G-code**
N/A

**Video**
https://youtu.be/OYFUG_z7dTA

leptun, 2021-08-15:
A few versions back this feature was changed so that it only opens the live adjust menu for the first few mm of Z. Afterwards, the live-adjust is not opened again as it doesn’t make sense to use it at that point.
It would also not make sense to have Move-Z opened during a print, so the long press is just ignored.

dalgibbard, 2021-08-15:
Oh that 100% makes sense! Thanks for the clarification.

closed 2021-08-15 by dalgibbard