prusa3d/Prusa-Firmware #3266: [QUESTION]<first layer calibration & mesh bed level>

ah1102, 2021-08-15:
Prusa mk3s
firmware 3.10
PINDA v2

**I noticed that my printed parts began to come off the edges. I calibrated the first layer and noticed that the zigzag on the left is worse than on the right.

For fun, I put two cut strips of A4 landscape sheet in the middle between the steel sheet and the bed. And I got the impression that my printer does not take into account the unevenness of the sheet. 

Are there any compensation limits?

It seemed to me that a printer with a PINDA sensor would take into account the unevenness of the surface. But when calibrating the first layer using a zigzag and visually without filament, I see that the print head does not adjust to the relief of the sheet.**

ah1102, 2021-08-16:
I installed firmware 3.8.0 and on it I also do not feel the work of the table alignment algorithm when calibrating the first layer. The extruder moves in a straight line and does not adjust to the surface relief. I feel a sheet of paper between the nozzle and the steel sheet.

In this photo, the edges have a distance of 0.06 mm, and the raised surface in the middle is 0.14 mm.

[image]

ah1102, 2021-08-18:
If I manually rotate, say, the right Z-bar in the bed renderer, it shows that the platform is tilted, but the print itself is no different, showing that the grid alignment is adjusted for the bed, but not for the ledges on it. construction plate.

JohnnyDeer, 2021-08-23:
Hi @ah1102, this issue is hardware related, please contact our Tech Support for troubleshooting and help. https://shop.prusa3d.com/en/ chat in right down corner. Link this issue to the chat please.
I´m closing issue here.

closed 2021-08-23 by JohnnyDeer