prusa3d/Prusa-Firmware #3267: [FEATURE REQUEST] Setup max. Z-High

andreasbica, 2021-08-16:
If it makes sense, enter what type of printer or upgrade the feature request applies to.
**Printer type** - [MINI+]
**Firmware** - 4.3.1

**Is your feature request related to a problem? Please describe.**
If I install the Sherpa Direct Extruder, the Motor is going to crash on the Z-Top and damage some parts.
  
**Describe the solution you'd like**
I prefer to set set the absolute Z-High Limit in: Menu > Settings

This improvement would be very helpful for some MINI+ users :)

wavexx, 2021-08-16:
Maybe you should log this to the Mini firmware repository?

andreasbica, 2021-08-16:
> Maybe you should log this to the Mini firmware repository?

Done :)
Thx for your notice ...

closed 2021-08-18 by JohnnyDeer