prusa3d/Prusa-Firmware #3270: [BUG] Exiting `Support` submenu like `Extruder Info` and others does not reset `lcd_timeoutToStatus` timer

gudnimg, 2021-08-21:
**Printer type** - MK3S
**Printer firmware version**-  3.10.1-RC1 and before

**MMU Upgrade** - N/A

**SD card or USB/Octoprint**
 SD card

**Describe the bug**

Minor bug which shouldn't be a huge concern except maybe to increase consistency.

When a user enters any of the submenus under `Support` like `Extruder Info` and then when the user exits the submenu the LCD will wait indefinitely in `Support` without user input. The expected behaviour would be after 30 seconds the printer returns to the main menu.

The bug here is `lcd_timeoutToStatus.stop()` is called and when `menu_back_if_clicked()` is called or `menu_back()`... the timer is not reset. This is another issue of when the LCD enters a menu, the timer is not set.
  
**To Reproduce**
1. Turn on printer
2. Click knob and scroll down and select the `Support` menu.
3. Scroll down and select `Extruder info`
4. Now click the knob to exit, be careful not to rotate the knob!
5. Wait for 30 seconds...
6. Observe that there is no change on the LCD.
  
**Expected behavior**
In step 6 above I would expect the LCD to return to the main menu.
 
**Note**

This is not an issue for `Sensor Info` but I think we should change the code

```cpp
void lcd_menu_show_sensors_state()                // NOT static due to using inside "Marlin_main" module ("manage_inactivity()")
{
	lcd_timeoutToStatus.stop();
	lcd_show_sensors_state();
	if(LCD_CLICKED)
	{
		lcd_timeoutToStatus.start();
		menu_back();
	}
}
```

to this below for consistency, but then this will be an issue too:

```cpp
void lcd_menu_show_sensors_state()                // NOT static due to using inside "Marlin_main" module ("manage_inactivity()")
{
	lcd_timeoutToStatus.stop();
	lcd_show_sensors_state();
	menu_back_if_clicked();
}
```

gudnimg, 2021-08-22:
Another note… even though clicking/pushing the knob to exit the submenu _does_ restart the timer, the submenu is executed once more afterwards which will stop the timer again.

leptun, 2021-08-22:
> Another note… even though clicking/pushing the knob to exit the submenu does restart the timer, the submenu is executed once more afterwards which will stop the timer again.

Not sure where this message went, but I’ve noticed some weird rendering issues with the sd menu when diving into a subdir where an empty menu is drown, then the items are sorted. Maybe it’s related.

gudnimg, 2021-08-23:
I deleted my comment because I thought for a moment what I said wasn't true. Turns out it is true, and I created this image below. I am thinking now if it is a good idea to check whether the menu changed and if it did, we reset the timer. That should cover the case where we go from the Status Menu and straight to the SD card menu when the SD card is inserted.

Thoughts?

[image]

gudnimg, 2023-11-02:
This issue is still present but I dont think it matters really. Time is better spent on other issues (like PrusaLink/Connect improvements)

closed 2023-11-02 by gudnimg