prusa3d/Prusa-Firmware #3274: [BUG] Printing in progress stops while printing is in progress with octoprint (V1.6.1).

beding, 2021-09-04:
Please, before you create a new bug report, please make sure you searched in open and closed issues and couldn't find anything that matches.

**Printer type** - [MK3S]
**Printer firmware version**-  [3.10.0]

**Octoprint [1.6.1]

**Describe the bug**
  Printing stops while not ready, nozzle remains hot against the object
  
**To Reproduce**
  Printing via prusa slicer a file that takes several hours.
  
**Expected behavior**
 Printing until ready.
  
**G-code**
[G.2_-_Knife_Guide_-_NEW_0.2mm_PETG_MK3S_1h5m.zip](https://github.com/prusa3d/Prusa-Firmware/files/7109604/G.2_-_Knife_Guide_-_NEW_0.2mm_PETG_MK3S_1h5m.zip)

  

**Video**

leptun, 2021-09-04:
Can you please upload also the serial logs from octoprint when the disconnect happened? As in the error that octoprint threw on disconnect (can probably be seen also in the terminal tab).

beding, 2021-09-05:
Hi Alex,

If the error is logged it should ne in here, it happened on 4-9-2021.

It happened also multiple times before I upgraded to octoprint 1.6.1  with firmware 3.10.0.

It never happened with firmware 3.9.2 the version I had before 3.10.0.

 

Regards,

Bert

wavexx, 2021-09-06:
@beding could you try with the latest 3.10.1-RC1 firmware?
We added an additional feature to debug hard problems such as this one.
If we're lucky, the printer will crash instead of just hanging, allowing us to see what is going on.

beding, 2021-09-06:
Thank you, I will do that.

What should I do when it happens?

Regards,

Bert

wavexx, 2021-09-06:
When the new FW crashes the data is saved internally. On the MK3/S this is fully automatic.
The data can be fetched after the crash using a new g-code: D21
See https://help.prusa3d.com/en/article/dump-memory-mk3-s-and-mk2-5-s_245237

It's not guaranteed to crash this way, although I definitely hope it does so. If you have a chance to record a video and/or happen to be there to see if there's anything else that could help understand what is going on it's always of great help.

beding, 2021-09-13:
Hi, I tested quit a lot this week.

I cannot reproduce the problem when debug print is ON in Octoprint.
So I continue now without it, but will have no serial logging then.

Regards,

Bert

closed 2023-11-13 by github-actions[bot]