prusa3d/Prusa-Firmware #3275: [FEATURE REQUEST] Support custom extruders like the 4.3.3-RC Mini Firmware

gregsaun, 2021-09-09:
**Printer type** - [e.g. MK3S+, MK3S, MK3, MK2.5S+, MK2.5S, MK2.5]

Supporting slightly different extruders can be a hard task for users, an easy way to set few parameters like it has been recently added to the Prusa Mini [here ](https://github.com/prusa3d/Prusa-Firmware-Buddy/releases/tag/v4.3.3-RC) would make this task eaiser, reduce the risks of user errors and open some possibilities for makers and developers. The idea wouldn't be to support "exotic" extruders but only allow some flexibilities for some parameters. This would also make sense as the developer is not available yet.
  
The solution would consist to turn some hardcoded values to variables we could change from the "secret" LCD menu. Here is a list of parameters I could imagine:
  - Save and return - while exiting this dialogue any adjusted value will be automatically saved.
  - Z-axis length - adjust the length of Z-axis based on your needs. In case you want to increase its length significantly, don't forget you might get limited by the cables.
  - Default Z length - the system will fall back to its default value.
  - Extruder direction - there are two choices of spin, “Prusa” and the second is obviously “Wrong”.
  - Default directions - set the direction back to the default (Prusa).

It could be interesting to add PINDA position but I understand this can be difficult/dangerous because of MBL.

matthew-humphrey, 2022-01-24:
Prusa - can you please give this some consideration? The open source nature of the printer and the availability of these mods is a big selling point of the printer. The change to support a different extruder top to nozzle height is a small one. If it helps, we could prepare a PR that would add some small code to enable an experimental menu option that enables a relaxed check on this dimension. This would mean that there would no longer be a need to fork the Prusa firmware and re-merge and test the changes with each new Prusa release.

natehale, 2022-01-24:
Prusa, please do this!
Thank you.

gourmandos, 2022-01-24:
Hi Prusa, fervent supporter of Greg's request, could you please implement the changes ?
Thank you

WarrenSchultz, 2022-01-24:
This would be a huge boon to the community.

marcolivierroy, 2022-01-25:
@DRracer would it be possible ?

jefflorenzini, 2022-01-25:
Please implement this.

orkolorko, 2022-01-25:
Voting for it!

DRracer, 2022-01-26:
Yes, it is obviously possible to implement, but in the 8bit FW we are fighting with every byte of CPUFLASH (and the new MMU is shaping up :wink: ). And especially UI interaction is hugely space-consuming. So, personally I'd vote against adding such a feature into the menu, but allow some simpler compilation procedure (extract the necessary configuration parameters somewhere) for users who run modded printers.

On the MINI, the situation was a bit more complicated due to FW signature (i.e. allowing ordinary users to tweak the parameters without breaking the appendix). On the 8bit platform there is no appendix, so users can build the FW on their own - or some community member can prepare modified builds for users who don't want to compile their own FW.

Does this approach make sense to you?

gregsaun, 2022-01-26:
@DRracer Thank you for your answer and I understand the problem with the flash capacity. Compilation procedure would at least make our work a little easier for sure and would be welcome. Do you think adding gcode for this would be possible / a good thing?

DRracer, 2022-01-26:
Adding a G-code + processing some of the parameters which are now hardcoded doesn't make much difference - the code will grow either way, because previously hardcoded instructions would have to read the values from elsewhere + the new G-code processing. Without actually trying I cannot say how much it will take, a rough estimate at least ~0.5KB.

So I still think making the parameters compile-time only is a win-win for all of us:
- still the same amount of code, just different instruction operands
- better configurability for modded printers

gregsaun, 2022-01-26:
Make sense for the g-code. At least a cleaning of hardcoded values would be helpful and maybe save some bytes for you as well.

- Could you confirm that it will also be possible to pass self test (and so wizard) with this new way?
- Do you think we could have a parameter for the AltFan as well (to have it disabled all the time)?

DRracer, 2022-01-27:
@gregsaun what you write makes sense - when there are `#define`s in the FW for various mods, it should cover all of the occasions including the selftest.
Disabling altfan can be done in a similar way (some `#ifdef`), but someone should look after the code and verify it still compiles and works without it. This is something we really can't support.

closed 2023-11-13 by github-actions[bot]