prusa3d/Prusa-Firmware #3276: [BUG] Filament runout issue

jlshelby, 2021-09-11:
Please, before you create a new bug report, please make sure you searched in open and closed issues and couldn't find anything that matches.

**Printer type** - [e.g. MK3S, MK3, MK2.5S, MK2.5, MK2S, MK2]
**Printer firmware version**-  [e.g. 3.8.1, 3.8.1-RC1, ...]
MK3S

**MMU Upgrade** - [e.g. MMU2S, MMU2, MMU1]
**MMU upgrade firmware version [e.g. 1.0.6, 1.0.6-RC2, ...]
MMU2S

**SD card or USB/Octoprint**
  Please let us know if you print via SD card or USB/Octoprint
SD card

**Describe the bug**
  A clear and concise description of what the bug is.
Filament 2 ran out of filament and stopped. I loaded the next spool and it started to resume but the MMU stopped feeding. The feeder from the heat element continued to feed the last bit from the first spool until it ran out but instead of detecting it had no more filament it just continued to print with no filament resulting in the rest of the print ruined.
  
**To Reproduce**
  Please describe steps to reproduce the behavior.
  
**Expected behavior**
  A clear and concise description of what you expected to happen.
  
**G-code**
  Please attach a G-code. This will make it easier for us to replicate the error.

**Video**
  Please attach a video. It usually helps to solve the problem.

jlshelby, 2021-10-06:
With the firmware in the ticket the filament will run out and continue printing without any detection. I have gone back to 3.7.2 (1.0.5) to resolve the issue.

gudnimg, 2022-12-20:
Adding to 3.13 milestone as a reminder to close this ticket. This issue should be already fixed in 3.13.

gudnimg, 2023-07-31:
Closing as this is now resolved with FW 3.13.0 and MMU FW 3.0.0 :)

closed 2023-07-31 by gudnimg