prusa3d/Prusa-Firmware #3277: [FEATURE REQUEST]limit sort by date to files in folder not folders themselves.

cnlson, 2021-09-18:
**Printer type** - [e.g. MK3S, MK3, MK2.5S, MK2.5, MK2S, MK2]
**MMU Upgrade** - [e.g. MMU2S, MMU2, MMU1]

I have probably 20 top level folders and several sub folders inside each. Right now if i choose sort by date time all of the folders sort by date time as well as the files. 

In my opinion only files should sort by date time. I would like my folders to always sort in alpha order so i can find them. 

For instance, if i save a file to my wife belinda's folder inside of halloween  and 2021 folders. I don't want to search through 20 top level to find belinda, i want to know it is in alpha order, similarly halloween and 2021 folders, Inside there, i want the files in newest to oldest format.

here is what happens now visually 

grandkids 3/2018
ring 2/2018
jeff folder 1/2018
misc folder 11-2017
belinda 10/2017
|-->  christmas 12/2018
        easter  4/2018
        st patties 3/2018
        halloween 10/2017
         |-->
               2021 1/2021
                |-->
                       files in newest to oldest

What i would like
belinda 10/2017
  |-->
        christmas 12/2018
        easter  4/2018
       halloween 10/2017
             2021
                }--> 
                       files in order. 
       st patties 3/2018
jeff folder 1/2018
misc folder 11-2017
ring 2/2018

The reason is that while drilling down, folders should be easy to find rather in a very random order, but files newest to oldest is better. 
  
**Describe the solution you'd like**
  A clear and concise description of what you want to happen.

gudnimg, 2021-09-19:
Do I understand correctly that you're asking for **only** folders to be sorted alphabetically? If so I want to suggest a more descriptive title like: "_Allow folders to be sorted differently than files_" :)

cnlson, 2022-01-03:
sorry for the delay, that is exactly what i am asking. I would like folders sorted by alpha and files sorted by date time

closed 2023-11-13 by github-actions[bot]