prusa3d/Prusa-Firmware #3278: [FEATURE REQUEST] Upgrade Marlin to 2.x

codemonkeyrawks, 2021-09-20:
Printer: MK3S, MK3

It would be helpful if Marlin 2.x (listed here: https://github.com/MarlinFirmware/Marlin) was pushed ahead. A feature that would also be helpful in relay would be to add several swap-able tool-heads to Marlin. 

Example:
+ PRUSA MK4-M ==> Mosquito Hotend (https://www.sliceengineering.com/collections/mosquito-the-professional-hotend)
+ PRUSA MK4-CW ==> Copperhead Wade Setup Hotend (https://gitlab.com/lulzbot3d/tool-heads/lutefisk/-/merge_requests/2)
+ PRUSA MK4-CB ==> Copperhead Bondtech (https://gitlab.com/lulzbot3d/tool-heads/lutefisk/-/merge_requests/5)
+ PRUSA MK4-HM ==> E3D Hemera (https://e3d-online.com/products/e3d-hemera-bowden-kit-1-75mm)

Feedback on XL:
Longer term the reason I like using my Taz 6 printer is (use case above) along with a larger print surface. I see their is development for the XL and feel like it should mock close to above and use parts like the Voron 2.x printer (https://www.formbot3d.com/products/voron-24-corexy-3d-printer-kit-with-different-print-sizes-for-choice) where as you can see users are not demanded to just using one tool design (the V6 is classic now). 

Additional Feedback on XL:
I will say it would be a nice requirement for the XL if it used this board (https://reprap.org/wiki/Rambo_v1.4) Rambo 1.4 over the Mini but does draw the question in actually needing a second extruder. Personally (I use one) but find this board to be useful if it was mounted under neath with a Meanwell power supply we would be great! This is what the Taz uses just would conform underneath and make it a slick printer!

codemonkeyrawks, 2021-09-20:
To follow up here is a few resources I ran across (looking for) however not all meet great requirements design like the Voron 2.x

* https://shop.lulzbot.com/sidekick - Toolhead Selector and Options
* https://shop.lulzbot.com/taz-pro?ref=KT-PR0050NA - This is one expensive printer! The idea of using a color phone screen in replace of the Voron 2.x (glcd) would be better. (https://www.digikey.com/en/products/detail/stmicro/B-LCD40-DSI1/6193106)
* https://www.digikey.com/en/products/detail/adafruit-industries-llc/4065/9919537 (Wifi Support)

Expected result I actually love Thomas! ==> (not the owner of the video) (https://www.youtube.com/watch?v=0E0dM0ZdpRE)

Again just sharing my ideas!

vintagepc, 2021-12-13:
Not speaking officially (just a volunteer contributor that's submitted some fixes here and there) but here's my 2 cents on your ideas just so it doesn't feel like you're being ignored:

Unfortunately, upgrading this firmware to 2.x is likely to be a major lift (on the scale of starting from scratch and re-writing every Prusa-specific feature and function they've added over the life of the MK3/S/+ - and then ensuring it works just as reliably as it does now without any compatibility-breaking changes). Essentially, a lot of work, with a lot of risk, for no net change. 

Additionally, in a practical sense the XL is unlikely to use the board you suggest as the 2560 is now very long in the tooth and there are more capable and cheaper offerings available now. Code space/RAM on AVR boards is a significantly limiting factor besides raw processing power and I don't think it'd be very future-proof to continue building on such a dated platform. Even the MIni is already on a 32 bit ARM platform and it'd be strange to take a step backwards from that now that they have people that are well-versed in developing for this hardware. (Not to mention the promo video shows an interface that looks a lot like the Mini... suggesting there's some shared code there)

closed 2023-11-13 by github-actions[bot]