prusa3d/Prusa-Firmware #3280: [QUESTION]<Enter comprehensive title> Erro compilar Arduino IDE Prusa mk2.5s

makerspacebauru, 2021-09-25:
Please, before you create a new question, please make sure you searched in open and closed issues and couldn't find anything that matches.

**What is your question?**

Friends could help me I'm compiling the prusa mk2.5s firmware file via ide arduino Rambo 1.3a I did the recording process as directed, but this is giving the error below, I'm using windows 10 and IDE Arduino 1.8.16.

Help me thanks.

wiring.co (símbolo do plugin): Na função __vector_23': (.text+0x0): multiple definition of __vector_23 '
sketch \ heatbed_pwm.cpp.o (símbolo do plugin) :(. text + 0x0): primeiro definido aqui
collect2.exe: erro: ld retornou 1 status de
saída status de saída 1
Erro compilando para placa RAMBo

leptun, 2021-09-25:
Please make sure you follow the instructions here: https://github.com/prusa3d/Prusa-Firmware#using-arduino
From the info you posted above it seems that you are using a newer arduino version and you are also using the wrong board definition. Prusa-firmware can't be compiled with the stock RAMBo board definition, you must use `Prusa Research AVR Boards by Prusa Research`. More info in the instructions above.

makerspacebauru, 2021-09-26:
ok thank you very much i will download arduino previous version and redo the process.

makerspacebauru, 2021-09-27:
> Certifique-se de seguir as instruções aqui: https://github.com/prusa3d/Prusa-Firmware#using-arduino
> Pelas informações que você postou acima, parece que você está usando uma versão mais recente do arduino e também está usando a placa errada definição. Prusa-firmware não pode ser compilado com a definição de placa RAMBo padrão, você deve usar `Prusa Research AVR Boards by Prusa Research`. Mais informações nas instruções acima.

 @leptun , good morning, I got a question, I'm using Rambo 1.3a board, prem in this process appears Select the target"Tools->Board->PrusaResearch Einsy RAMBo", the settings of this board are different from Mini Rambo1.3a.**

leptun, 2021-09-27:
The arduino board definition for the Einsy rambo and the Mini rambo is one and the same. What will make a difference is the variant you select (the config you use as Configuration_prusa.h)

3d-gussner, 2021-09-29:
@makerspacebauru Good morning,

I just:
- downloaded the firmware from https://github.com/prusa3d/Prusa-Firmware/tree/MK3_3.10.1
- Downloaded Arduino IDE 1.18.6
- added the https://raw.githubusercontent.com/prusa3d/Arduino_Boards/master/IDE_Board_Manager/package_prusa3d_index.json to the boards manager
- Installed **Prusa Research AVR Boards** version `1.0.4`
- Copied the `Firmware/variants/1_75mm_MK25S-RAMBo13a-E3Dv6full.h` to `Firmware/Configuration_prusa.h`
- Opened Arduino IDE 1.18.6 and the `Firmware.ino`
- Changed in `config.h` to use primary language only.

```
  //LANG - Multi-language support
#define LANG_MODE              0 // primary language only
//#define LANG_MODE              1 // sec. language support
```

- Changed to Prusa board in Arduino **Tools->Board** to
  - `Prusa Research AVR Boards`
    - `PrusaResearch Einsy RAMBo`

- Compiled the Firmware
```
Sketch uses 205498 bytes (80%) of program storage space. Maximum is 253952 bytes.
Global variables use 5955 bytes of dynamic memory.
```

So it looks fine.

Maybe a better way is that you try `./PF-build.sh` to compile the firmware see https://github.com/prusa3d/Prusa-Firmware/tree/MK3_3.10.1#using-linux-subsystem-under-windows-10-64-bit
and https://github.com/prusa3d/Prusa-Firmware/tree/MK3_3.10.1#using-git-bash-under-windows-10-64-bit

If you have you can setup a Ubuntu Virtual Machine it might work even better.

- It prepares a supported build environment for you
- And makes multi language firmware.

makerspacebauru, 2021-09-29:
> @makerspacebauru Bom dia,
> 
> Eu só:
> 
> * baixou o firmware de https://github.com/prusa3d/Prusa-Firmware/tree/MK3_3.10.1
> * Download do Arduino IDE 1.18.6
> * adicionou https://raw.githubusercontent.com/prusa3d/Arduino_Boards/master/IDE_Board_Manager/package_prusa3d_index.json ao gerenciador de placas
> * Versão instalada das **placas Prusa Research AVR**`1.0.4`
> * Copiou `Firmware/variants/1_75mm_MK25S-RAMBo13a-E3Dv6full.h`para`Firmware/Configuration_prusa.h`
> * Arduino IDE 1.18.6 aberto e o `Firmware.ino`
> * Alterado `config.h`para usar apenas o idioma principal.
> 
> ```
>   //LANG - Multi-language support
> #define LANG_MODE              0 // primary language only
> //#define LANG_MODE              1 // sec. language support
> ```
> 
> * Alterado para placa Prusa em **Ferramentas Arduino-> Placa** para
>   
>   * `Prusa Research AVR Boards`
>     
>     * `PrusaResearch Einsy RAMBo`
> * Compilou o Firmware
> 
> ```
> Sketch uses 205498 bytes (80%) of program storage space. Maximum is 253952 bytes.
> Global variables use 5955 bytes of dynamic memory.
> ```
> 
> Então, parece bom.
> 
> Talvez a melhor maneira é tentar `./PF-build.sh`compilar o firmware, consulte https://github.com/prusa3d/Prusa-Firmware/tree/MK3_3.10.1#using-linux-subsystem-under-windows-10-64-bit e [https : //github.com/prusa3d/Prusa-Firmware/tree/MK3_3.10.1#using-git-bash-under-windows-10-64-bit](https://github.com/prusa3d/Prusa-Firmware/tree/MK3_3.10.1#using-git-bash-under-windows-10-64-bit)
> 
> Se você tiver, pode configurar uma Máquina Virtual Ubuntu, ela pode funcionar ainda melhor.
> 
> * Ele prepara um ambiente de construção compatível para você
> * E faz firmware multilíngue.

 @3d-gussner 3d-gussner Good morning friend, our time zones are not from Brazil, I did the whole process again now this error appears below follows.

"C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avr-gcc" -Os -g -flto -Wl,--gc-sections,--relax -mmcu=atmega2560 -Wl,-u,vfprintf -lprintf_flt -lm -Wl,--script=C:\Users\Samuel Conegunes\AppData\Local\Arduino15\packages\PrusaResearch\hardware\avr\1.0.4/ldscripts/avr6.xn -o "C:\Users\SAMUEL~1\AppData\Local\Temp\arduino_build_629880/Firmware.ino.elf" "C:\Users\SAMUEL~1\AppData\Local\Temp\arduino_build_629880\sketch\adc.c.o" "C:\Users\SAMUEL~1\AppData\Local\Temp\arduino_build_629880\sketch\bootapp.c.o" "C:\Users\SAMUEL~1\AppData\Local\Temp\arduino_build_629880\sketch\language.c.o" "C:\Users\SAMUEL~1\AppData\Local\Temp\arduino_build_629880\sketch\messages.c.o" "C:\Users\SAMUEL~1\AppData\Local\Temp\arduino_build_629880\sketch\pat9125.c.o" "C:\Users\SAMUEL~1\AppData\Local\Temp\arduino_build_629880\sketch\rbuf.c.o" "C:\Users\SAMUEL~1\AppData\Local\Temp\arduino_build_629880\sketch\sm4.c.o" "C:\Users\SAMUEL~1\AppData\Local\Temp\arduino_build_629880\sketch\spi.c.o" "C:\Users\SAMUEL~1\AppData\Local\Temp\arduino_build_629880\sketch\swi2c.c.o" "C:\Users\SAMUEL~1\AppData\Local\Temp\arduino_build_629880\sketch\timer02.c.o" "C:\Users\SAMUEL~1\AppData\Local\Temp\arduino_build_629880\sketch\tone04.c.o" "C:\Users\SAMUEL~1\AppData\Local\Temp\arduino_build_629880\sketch\twi.c.o" "C:\Users\SAMUEL~1\AppData\Local\Temp\arduino_build_629880\sketch\uart2.c.o" "C:\Users\SAMUEL~1\AppData\Local\Temp\arduino_build_629880\sketch\xflash.c.o" "C:\Users\SAMUEL~1\AppData\Local\Temp\arduino_build_629880\sketch\AutoDeplete.cpp.o" "C:\Users\SAMUEL~1\AppData\Local\Temp\arduino_build_629880\sketch\BlinkM.cpp.o" "C:\Users\SAMUEL~1\AppData\Local\Temp\arduino_build_629880\sketch\Configuration.cpp.o" "C:\Users\SAMUEL~1\AppData\Local\Temp\arduino_build_629880\sketch\ConfigurationStore.cpp.o" "C:\Users\SAMUEL~1\AppData\Local\Temp\arduino_build_629880\sketch\Dcodes.cpp.o" "C:\Users\SAMUEL~1\AppData\Local\Temp\arduino_build_629880\sketch\Firmware.ino.cpp.o" "C:\Users\SAMUEL~1\AppData\Local\Temp\arduino_build_629880\sketch\MarlinSerial.cpp.o" "C:\Users\SAMUEL~1\AppData\Local\Temp\arduino_build_629880\sketch\Marlin_main.cpp.o" "C:\Users\SAMUEL~1\AppData\Local\Temp\arduino_build_629880\sketch\Sd2Card.cpp.o" "C:\Users\SAMUEL~1\AppData\Local\Temp\arduino_build_629880\sketch\SdBaseFile.cpp.o" "C:\Users\SAMUEL~1\AppData\Local\Temp\arduino_build_629880\sketch\SdFatUtil.cpp.o" "C:\Users\SAMUEL~1\AppData\Local\Temp\arduino_build_629880\sketch\SdFile.cpp.o" "C:\Users\SAMUEL~1\AppData\Local\Temp\arduino_build_629880\sketch\SdVolume.cpp.o" "C:\Users\SAMUEL~1\AppData\Local\Temp\arduino_build_629880\sketch\Servo.cpp.o" "C:\Users\SAMUEL~1\AppData\Local\Temp\arduino_build_629880\sketch\Timer.cpp.o" "C:\Users\SAMUEL~1\AppData\Local\Temp\arduino_build_629880\sketch\backlight.cpp.o" "C:\Users\SAMUEL~1\AppData\Local\Temp\arduino_build_629880\sketch\cardreader.cpp.o" "C:\Users\SAMUEL~1\AppData\Local\Temp\arduino_build_629880\sketch\cmdqueue.cpp.o" "C:\Users\SAMUEL~1\AppData\Local\Temp\arduino_build_629880\sketch\conv2str.cpp.o" "C:\Users\SAMUEL~1\AppData\Local\Temp\arduino_build_629880\sketch\eeprom.cpp.o" "C:\Users\SAMUEL~1\AppData\Local\Temp\arduino_build_629880\sketch\first_lay_cal.cpp.o" "C:\Users\SAMUEL~1\AppData\Local\Temp\arduino_build_629880\sketch\fsensor.cpp.o" "C:\Users\SAMUEL~1\AppData\Local\Temp\arduino_build_629880\sketch\heatbed_pwm.cpp.o" "C:\Users\SAMUEL~1\AppData\Local\Temp\arduino_build_629880\sketch\la10compat.cpp.o" "C:\Users\SAMUEL~1\AppData\Local\Temp\arduino_build_629880\sketch\lcd.cpp.o" "C:\Users\SAMUEL~1\AppData\Local\Temp\arduino_build_629880\sketch\menu.cpp.o" "C:\Users\SAMUEL~1\AppData\Local\Temp\arduino_build_629880\sketch\ [... cut ...]
# - from numeric vendor ID, set to Unknown otherwise
build.usb_manufacturer="Unknown"
build.usb_flags=-DUSB_VID={build.vid} -DUSB_PID={build.pid} '-DUSB_MANUFACTURER={build.usb_manufacturer}' '-DUSB_PRODUCT={build.usb_product}'

____________________________________________________________________________________________________________________________

I have another machine with Linux Mint I will try to load it, I am starting up, I have some difficulties but I really want to get this printer to happen. Tanks

makerspacebauru, 2021-09-30:
@3d-gussner @leptun 
Friends, when finalizing the compilation for the board, this error appeared, as we can adjust, I believe there is little adjustment to make this printer work.

avrdude: stk500v2_setparm(): failed to set parameter 0x9e
avrdude: stk500v2_command(): unknown status 0xc8
avrdude: initialization failed, rc=-1
         Double check connections and try again, or use -F to override
         this check.

avrdude: stk500v2_command(): unknown status 0x01
avrdude: stk500v2_disable(): failed to leave programming mode
Ocorreu um erro enquanto o sketch era carregado

another LCD dot turned on but nothing appears on the screen or will only appear if the firmware flashes.

leptun, 2021-09-30:
Does flashing an original firmware via PrusaSlicer work?

makerspacebauru, 2021-10-01:
> A atualização de um firmware original via PrusaSlicer funciona?

@leptun, Good morning, sorry to answer just today, friend I'm trying to upload via ide but now this error appears, you can also do it via slicer but the board for being Chinese doesn't come with Bootloader Prusa, to do this process it would give but not I found it as a teacher to do, I found it only for Eynsi Rambo, friend I don't know what I do anymore.
Below is a very detailed link that teaches the process that guided me, below I send the error when saving to the board.

________________________________________________________________________________________________________________

avrdude: stk500v2_setparm(): failed to set parameter 0x9e
avrdude: stk500v2_command(): unknown status 0xc8
avrdude: initialization failed, rc=-1
         Double check connections and try again, or use -F to override
         this check.

avrdude: stk500v2_command(): unknown status 0x01
avrdude: stk500v2_disable(): failed to leave programming mode
Ocorreu um erro enquanto o sketch era carregado

3d-gussner, 2021-10-02:
@makerspacebauru So you don't have a Ultimachine or Prusa Einsy correct?
Please read https://github.com/prusa3d/Prusa-Firmware/issues/910#issuecomment-406855786
If you get this working you can upload the Prusa-bootloader from the Arduino IDE

makerspacebauru, 2021-10-02:
> @makerspacebauru Então você não tem Ultimachine ou Prusa Einsy correto? Por favor, leia [# 910 (comentário).](https://github.com/prusa3d/Prusa-Firmware/issues/910#issuecomment-406855786) Se você conseguir fazer isso funcionar, você pode fazer o upload do carregador de inicialização Prusa a partir do IDE do Arduino

@3d-gussner Hello friend, the board I bought in china mini Rambo 1.3a, the process for Bootloader [and even from Eynsi I'm having difficulties to understand, it's well explained for Eynsi but it doesn't have for the mini Rambo, well you already gave me the way I'll see there is Rambo mini scheme, thank you.

3d-gussner, 2021-10-02:
Have a look here, you should find the correct pins

closed 2023-11-14 by github-actions[bot]