prusa3d/Prusa-Firmware #3283: [BUG] MMU2S Selector Logic Issue after Reset & Recovery

shintaroaibara, 2021-10-05:
**Printer type** - MK3S
**Printer firmware version**-  3.10.0 

**MMU Upgrade** - MMU2S
**MMU upgrade firmware version 1.0.6

**SD card or USB/Octoprint**
Octoprint

**Describe the bug**
I have a reproducible issue that if my MMU2S has an issue and stops (All Red/Green LEDs blinking), and I resolve the issue and hit the reset button, the first thing that happens is that the MMU2S resets and idles with the green LED lit on position 1. This is the case even if the last filament position was not position 1, and if the print is restarted, it will start loading position 1. To circumvent this issue, after the reset I can move the selector manually using the right button of the MMU2S, and it is able to go to the correct selector position and resume printing.

However, after that layer is done, and for the next tool change, somehow the MMU2S forgets that it needs to unload, and will trigger the selector to move even if there is a filament still in the system. So what happens is that it pulls the filament (which snaps, as it is PLA), and then there is a load failure. Interestingly, if I then intervene and remove the snapped filament, and restart the loading procedure (which doesn't need a reset) then the MMU2S will remember to unload the next filament.

Therefore it feels like for some reason the reset button makes the MMU2S forget that it needs to unload after that particular tool.

**To Reproduce**
Cause a MMU2S issue (idler tension?) - All Red/Green LEDs blinking
Fix the issue (e.g. reset Idler position and screw tension)
Hit the MMU2S reset button
Observe that a Green LED light is lit on position 1
Move the selector to previous position using the Right button of the MMU2S
Restart Print
Wait until next tool change
Issue occurs here, where the selector will move without unloading the filament
  
**Expected behavior**
MMU2S should unload the filament before moving the selector

gudnimg, 2023-06-22:
Adding this to 3.13 milestone, I think this issue should be closed once FW 3.13 is released.

gudnimg, 2023-07-31:
Closing as this is now resolved with FW 3.13.0 and MMU FW 3.0.0 :)

closed 2023-07-31 by gudnimg