prusa3d/Prusa-Firmware #3284: [BUG] TMC2130_SERVICE_CODES_M910_M918 breaks compiling

KrX3D, 2021-10-05:
Please, before you create a new bug report, please make sure you searched in open and closed issues and couldn't find anything that matches.

**Printer type** - MK3S
**Printer firmware version**- Prusa-Firmware-3.10.1-RC1

**Describe the bug**
When i uncommenc "#define TMC2130_SERVICE_CODES_M910_M918" in Configuration_prusa.h i get this error
  
[image]

gudnimg, 2021-12-06:
Fix was merged into 3.10.1 branch: https://github.com/prusa3d/Prusa-Firmware/pull/3304

gudnimg, 2023-06-22:
As this was fixed in 3.10.1, 2 years ago I'm closing this issue.

closed 2023-06-22 by gudnimg