prusa3d/Prusa-Firmware #3287: [QUESTION] 3.10.1 Release Date?

Danafra, 2021-10-06:
Sorry for such an elementary question, but has there been any discussion as to when this will be released?

For that matter, is there a way one can check to determine progress to release here?

jiri-jirus, 2021-12-02:
RC1 is not able to connect via OctoPrint on my printers. Did anyone suceed with this?

marcolivierroy, 2021-12-03:
> 

It works with mine. Make sure you put the right config for the port you are using. (USB Std B or Header Socket)

jiri-jirus, 2021-12-17:
> > 
> 
> It works with mine. Make sure you put the right config for the port you are using. (USB Std B or Header Socket)

well this seems to be related to new SN reading code

3d-gussner, 2021-12-17:
@Danafra It has been released :partying_face:  please close the question.

closed 2021-12-20 by Danafra