prusa3d/Prusa-Firmware #3294: [BUG]<Octoprint not printing. In 4.3.3>

br3jlm, 2021-10-16:
Please, before you create a new bug report, please make sure you searched in open and closed issues and couldn't find anything that matches.

**Printer type** - [e.g. MK3S, MK3, MK2.5S, MK2.5, MK2S, MK2]
**Printer firmware version**-  [e.g. 3.8.1, 3.8.1-RC1, ...]

**MMU Upgrade** - [e.g. MMU2S, MMU2, MMU1]
**MMU upgrade firmware version [e.g. 1.0.6, 1.0.6-RC2, ...]

**SD card or USB/Octoprint**
  I can t print from USB but not from octoprint

**Describe the bug**
  A clear and concise description of what the bug is.
  
**To Reproduce**
  Please describe steps to reproduce the behavior.
  
**Expected behavior**
  A clear and concise description of what you expected to happen.
  
**G-code**
  Please attach a G-code. This will make it easier for us to replicate the error.

**Video**
  Please attach a video. It usually helps to solve the problem.

leptun, 2021-10-16:
@gudnimg that’s just the generic issue template. The title was changed to “Operator error”, so I assume this is an accidental issue report.

br3jlm, 2021-10-16:
> We will need more information that this. You are running old firmware (3.8.1), could you please try 3.10.0 or 3.10.1-RC1?
>
> What have you tried to do to solve this? Do you have any serial logs, error logs?
>
> What printer are you using? Does it have a MMU?
>
> Expected behavior
> I didn't realize you can't use filament swap with octoprint￼
>
> How is this related to your issue?

Hello.  

I  am using a mini+ running 4.3.3.   I was trying to print a color swap G-Code to the mini plus threw octoprint and I didn’t know you can’t to that( I wouldn't work).  
I ran another g code with color swap and went to the printer just fine 

Regards
Justin

br3jlm, 2021-10-16:
I didn’t know you couldn’t delete things.   I will do more research first next time

3d-gussner, 2021-11-15:
@br3jlm Can you please close this issue?

gudnimg, 2023-07-24:
Closing this Github issue as it's likely resolved but user maybe forgot to close it.

closed 2023-07-24 by gudnimg