prusa3d/Prusa-Firmware #3300: [BUG]<Enter comprehensive title>

Stiixxy, 2021-11-11:
**Printer type** 
MK3
**Printer firmware version**
3.10.0

**Describe the bug**
Fillament sensor is not used during first layer calibration

**To Reproduce**
Cut fillament off shortly above extruder and start first layer calibration. Once it runs out the calibration continues
  
**Expected behavior**
First layer calibration should at the very least stop and move the extruder up.

pawelzwronek, 2021-11-14:
My very first layer calibration started even without filament 🤔

3d-gussner, 2021-11-15:
@Stiixxy Can you please edit your Issue and add a better title than `[BUG]<Enter comprehensive title>` 

Why would you even try a 1st layer cal without filament OR if you see that the filament is too short OR cut it during 1st layer calibration?

There are some checks/features disabled during 1st layer calibration to insure best chance of getting it right in a repeatable way.

I am not sure enabling/checking the FRS during 1st layer cal should be priority to fix as the user sees if there is filament loaded or not.

Stiixxy, 2021-11-15:
@3d-gussner  whoops, sorry missed the title

I do agree FRS isn't really a priority, as no one leaves their printer alone during this calibration.

For me, my fillament got tangled (accidentally pulled it though the spool), i had to cut it. For me this was during the setup wizard of my new printer. Tbh, i hoped the fillament sensor would kick in so i could reload it and start again. In the end i resetted the printer and started the wizard again.

So in the end i do agree, it's not a major priority bug. But i thought it might be worth reporting

3d-gussner, 2021-12-07:
@Stiixxy Please consider to close this issue.

closed 2021-12-07 by Stiixxy