prusa3d/Prusa-Firmware #3302: [QUESTION] PT1000 Temp sensor, Max temp Limit

mop999x, 2021-11-14:
I am trying to Upgrade a temp sensor on my mk3s I was wondering for the PT1000 it’s a 500c, does prusa limit the temp to 300c for it in the firmware ?  https://e3d-online.com/products/pt1000-temperature-sensor

https://github.com/prusa3d/Prusa-Firmware/blob/MK3/Firmware/thermistortables.h

#if (THERMISTORHEATER_0 == 1047) || (THERMISTORHEATER_1 == 1047) || (THERMISTORHEATER_2 == 1047) || (THERMISTORBED == 1047) // Pt1000 with 4k7 pullup
const short temptable_1047[][2] PROGMEM = {
// only few values are needed as the curve is very flat  
  PtLine(0,1000,4700)
  PtLine(50,1000,4700)
  PtLine(100,1000,4700)
  PtLine(150,1000,4700)
  PtLine(200,1000,4700)
  PtLine(250,1000,4700)
  PtLine(300,1000,4700)
};
#endif

or is this just used to describe a mathematical linear/curve of temp to resistant ratio?
also where are the temp table values for this sensor acquired from, link?

3d-gussner, 2021-11-16:
Before I begin just a 

**WARNING**:

**The stock Prusa MK3S uses E3D V6 with an aluminum heater block, brass nozzle and stainless steel heatbreak so don't try to go above the set limits without changing the hotend and other parts!
The Extruder body and Fan duct in PETG/ABS may not withstand these temperatures!**

Here a link you can get more information: https://e3d-online.zendesk.com/hc/en-us/articles/360016227698-V6-Maximum-Temperature-Upgrade-Path-
https://www.heraeus.com/media/media/hne/datasheets/m_sensors/en_10/m_222_e.pdf

You find most things the variants files:
- https://github.com/prusa3d/Prusa-Firmware/blob/MK3/Firmware/variants/1_75mm_MK3S-EINSy10a-E3Dv6full.h#L323-L328 for the MAXTEMP
- https://github.com/prusa3d/Prusa-Firmware/blob/MK3/Firmware/variants/1_75mm_MK3S-EINSy10a-E3Dv6full.h#L323-L328 for PID
- https://github.com/prusa3d/Prusa-Firmware/blob/MK3/Firmware/variants/1_75mm_MK3S-EINSy10a-E3Dv6full.h#L589-L595 which thermistor is used

In thermistorstable.h you find the right table for the PT1000 https://github.com/prusa3d/Prusa-Firmware/blob/MK3/Firmware/variants/1_75mm_MK3S-EINSy10a-E3Dv6full.h#L589-L595
You may have a look in vanilla Marlin https://github.com/MarlinFirmware/Marlin/blob/596b3c761459737fdbec529fd35abbf45a11b82b/Marlin/src/module/thermistor/thermistor_1047.h to update the Prusa-Firmware values for higher temperatures.

Please be very careful with these temperatures and DO NOT exceed the material limits:
- heater block
- heatbreak
- nozzle
- plastic parts nearby

I even doubt that you will be able to print that high temperature materials without highly modifying your printer.

mop999x, 2021-11-21:
Ya most i do have most of the high temp parts nozzle X, tungsten carbide nozzle, copper block, titanium break, High Precision Heater Cartridges just need to install them

i am a bit unsure about the petg parts handling high temp though. not sure about atmospheric or environment temp that would spread from the nozzle to plastic part how that would work or cause the parts to melt, if printing at 500c might be bad but maby lower high temp caped at 350c or 400c if that would cause problems too.

3d-gussner, 2021-12-07:
@mop999x If your question has been answered, please consider to close this issue.
It really helps if these don't stay open and have to be reviewed so Prusa is forced to close them.

3d-gussner, 2021-12-17:
@JohnnyDeer As there is no reaction from @mop999x please close this issue.
[stale]

closed 2023-11-14 by github-actions[bot]