prusa3d/Prusa-Firmware #3308: [BUG] G92 allows printer to exceed physical limits

haarp, 2021-11-29:
**Printer type** - MK3
**Printer firmware version** -  3.10

**MMU upgrade** - no

**SD card or USB/Octoprint** - irrelevant

**Describe the bug**
When G92 is used to set new coordinates/workspace, the printer may now exceed its physical limits, potentially causing parts to crash and grind.
  
**To Reproduce**
- Home printer
- Move Y to 200
- G92 Y0
- Move Y to 20
- Observe Y crashing into endstop
  
**Expected behavior**
The printer should refuse to do the last movement. According to [Marlin notes](https://marlinfw.org/docs/gcode/G092.html), the printer should maintain physical boundaries, even when a different workspace was set using G92.

Additonally, neither G53 or G92.1 are available to reset the printer to native coordinates, making this bug hard to workaround.

Thanks!

3d-gussner, 2021-12-07:
@haarp Prusa Firmware has been forked from marlin 1.0.x and frankly (my opinion) drifted far away from the roots.

Whatever:
If you use `G92` manually you should also ensure that you stay within the hardware limits.
I am not aware of any Slicer that will generate gcode files including `G92` on x,y,z and "force" a movement outside the defined boundaries.

This is more a feature request and feel free to create a PR to improve the "rarely" used way you describe.

leptun, 2021-12-07:
I personally prefer being able to go outside machine limits with G92. This is the only way to lower the Z axis when Z is not homed. Not sure how you could do this in marlin.

jiri-jirus, 2021-12-17:
> 

in Marlin this is done by temporarily turning off Software Endstops (M221 S0)

closed 2023-11-14 by github-actions[bot]