prusa3d/Prusa-Firmware #3309: [FEATURE REQUEST] Interacting with LCD via serial/GCODE

przemo-c, 2021-12-01:
When printing remotely we can't interact with prompts on LCD That can block remote print job on crash  etc.

We need to have either extensive action comand support enabling to respond to any on screen prompt remotely or
GCODE that sends over serial contents of the LCD and commands for operating the knob and reset button.

bmuessig, 2021-12-08:
On my previous job, we worked on exactly this problem on the MK3 in order to create a mostly remote controlled print farm

bmuessig, 2021-12-08:
It is possible to hook into the LCD driver and to signal updates to the PC by using the characters 0x80 to 0xff in order to create a second "interleaved" serial stream that carries display data and button presses in the respective direction.
Since we had our own custom Go software to communicate with the printer, we could just separate the streams when reading them.

bmuessig, 2021-12-08:
It would have surely also worked to silence all existing serial output, but we did not want to do this. As the debug output on the MK3S is always ASCII, the 8th character bit is left unused, which we can re-purpose as an extended command flag (a bit similar to how UTF-8 is implemented).

leptun, 2021-12-08:
https://photos.app.goo.gl/tAMGANgmKMPW1fTDA
Just a random thing I did 2 years ago. Obviously not perfect, but printing the display to serial was just a side effect of what I was trying to implement.

przemo-c, 2021-12-17:
It wouldn't even have to be realtime. Just upon request as a response to command. Not sure how it's structured but if there's a way to print out what current lcd display should be displaying(redraw screen to serial)  as a responce to command and using either the same command with parameter to interact as in simulate knob turn/press and reset. 

That would solve issues with crash detection pauses and acknowledging thermal runaway protection and just a troubleshooting reset of the printer.

bmuessig, 2021-12-17:
If you need a command to print a full frame and since you can't really read out the LCD, you have to maintain a screen buffer to which to copy any changes that go to the LCD. This buffer could then be printed via G-Code. It would also be possible to send button input via G-Code by injecting forced keypresses and scrolls.

przemo-c, 2021-12-17:
> If you need a command to print a full frame and since you can't really read out the LCD, you have to maintain a screen buffer to which to copy any changes that go to the LCD. This buffer could then [...]

That's what I've figured. sort of redraw to serial rather than keep up to date copy or send over serial on each lcd update.

bmuessig, 2023-11-06:
Bump as still needed

3d-gussner, 2024-03-08:
This won't be implemented except a volunteer is willing to provide a pull request.

bmuessig, 2024-03-08:
I’d be happy to implement it @3d-gussner

3d-gussner, 2024-03-08:
Great looking forward to review it. Please be careful with the resources as resource hungry PRs need a lot of good arguments to be considered.

bmuessig, 2024-05-20:
I started a new job and didn’t have much time recently. I am still going to implement this feature soon.

3d-gussner, 2024-05-20:
@bmuessig Looking forward to see your implementation, thanks for the update.

Tirpitz93, 2024-07-16:
Which GCODES are we thinking for this feature?
I am thinking for example:
`M940` to echo the contents of the LCD
`M941` for sending input events with params `E` for encoder input with integer increment or decrement. `B` for button event. `N` long press. 
And possibly `U`, `D` `L`, `R` for direction keys if present or required.

bmuessig, 2024-07-16:
@Tirpitz93 That would work. When I started working on the code, I also began implementing basic partial refresh and worked on converting the international characters to ASCII.

Tirpitz93, 2024-07-19:
> I would suggest `M941` to send the delta since the last full frame read with `M940` and have `M942` as an input injection mechanism.

My implementation would not track changes to reduce memory footprint.
I think the memory footprint is more important than saving a few chars when requesting the LCD contents.

bmuessig, 2024-07-19:
@Tirpitz93 We only need one bit per character to store a change flag, which we have free as we’re only using ASCII replacement characters

leptun, 2024-07-19:
Just fyi we use all 8 bits on the lcd because of diacritics. Those are mapped from 0x80 onwards

Tirpitz93, 2024-07-19:
My LCD buffer array is (LCD_WIDTH+1) * LCD_HEIGHT chars to give a null terminated string for each line.
I guess we could remove the terminator too, and send LCD_WIDTH characters each time.
And the previous comment basically addresses the question of unicode characters.

bmuessig, 2024-07-19:
There are no Unicode characters in the Prusa firmware. It is using a custom encoding that utilizes the custom characters on the printer‘s LCD controller. My suggestion is to use the ASCII replacement characters for international characters already included in the firmware, for our character buffer. This gives us one bit to spare as a change flag. We then send consecutive changes regularly, prefixed by their start index in the buffer.

przemo-c, 2024-09-19:
It's still a needed feature.

bmuessig, 2024-11-26:
Bump to keep open

bmuessig, 2025-01-27:
Bump to keep open, will have some time starting in two weeks

Tirpitz93, 2025-08-11:
I have been working on this intermittently.