prusa3d/Prusa-Firmware #3314: [BUG]Build fails if `TEMP_STAT_LEDS` is enabled

gudnimg, 2021-12-10:
**Describe the bug**
If `TEMP_STAT_LEDS` preprocessor definition is enabled in `Configuration.h` the build will fail. This is because the two preprocessor symbols `STAT_LED_RED` and `STAT_LED_BLUE` are not defined anywhere in the project. I suspect this is some leftover code from Marlin? We can potentially remove this code alltogether to reduce code *noise*.

[image]

  
**To Reproduce**
Enable `TEMP_STAT_LEDS` preprocessor definition.
  
**Expected behavior**
Build succeeds without any warnings or errors.

JohnnyDeer, 2021-12-13:
Hi @gudnimg, thank you for reporting. We are aware of this bug, unfortunately at this moment have low priority. We will fix it in one of the future releases. Thank you for understanding and patience.

gudnimg, 2021-12-25:
@JohnnyDeer I created a pull request https://github.com/prusa3d/Prusa-Firmware/pull/3327 that will resolve this issue.

closed 2022-02-07 by 3d-gussner