prusa3d/Prusa-Firmware #3319: [BUG]Replacement non-genuine Einsy board timeouts connecting to OctoPrint from 3.10.1-rc onwards

jiri-jirus, 2021-12-17:
**Printer type** - MK3S
**Printer firmware version** -  3.10.1-rc and newer

**SD card or USB/Octoprint**
Octoprint 1.7.2

**Describe the bug**
Replacement non-genuine Einsy board timeouts connecting to OctoPrint from 3.10.1-rc onwards. Simply this is because non-genuine board does not have CZPX...... serial number, and new code that harvests SN in CZPX format from AT90 prevents OctoPrint from connecting.
  
**To Reproduce**

  
**Expected behavior**
non-genuine boards are able to connect to OctoPrint.

adelyser, 2022-01-02:
I ran into this issue, with an authentic MK3S with a real SN. Apparently my EEPROM got corrupted and my serial number was shiftted about 5 bytes. I could no longer connect to fix things. Here's how I ultimately fixed it:

1. Revert to 3.10.0 firmware
2. Re-write the Serial number to eeprom using this command (this one will write CZPX11111111111111)
`D3 Axd15 C20 X435a5058313131313131313131313131313131`
3. Re-flashed to 3.10.1 firmware

With a non-genuine board, you could just put whatever serial you want in there using the above steps.

This is certainly an issue, and it would be nice if there was a way to continue connecting to the board to fix it, and possibly some sort of notification to the user there's an issue?

leptun, 2022-01-18:
@jiri-jirus Can you please send a serial log from octoprint where the failure can be observed? I want to see how the exchange happens at the beginning.

jiri-jirus, 2022-02-19:
> 

Hello, there is serial log from Octoprint: 

`Changing monitoring state from "Offline" to "Opening serial connection"
Connecting to port /dev/ttyUSB.MK3-01, baudrate 115200
Changing monitoring state from "Opening serial connection" to "Connecting"
Connected to: Serial<id=0x6d59c5d0, open=True>(port='/dev/ttyUSB.MK3-01', baudrate=115200, bytesize=8, parity='N', stopbits=1, timeout=10.0, xonxoff=False, rtscts=False, dsrdtr=False), starting monitor
Send: N0 M110 N0*125
Recv: Command not found!
Recv: start
Send: N0 M110 N0*125
Recv: ;S
Recv: ;S
Recv: ;S
Recv: ;S
Recv: ;S
Recv: SN update failed
Recv: echo: 3.10.1-RC1-5389
Recv: Commit Hash: 93e777d8
Recv: echo: Last Updated: Jul 16 2021 21:31:57 | Author: (3D-Gussner, Caribou3d)
Recv: Compiled: Jul 16 2021
Recv: echo: Free Memory: 2041  PlannerBufferBytes: 1760
Recv: echo:Stored settings retrieved
Recv: adc_init
Recv: Extruder fan type: NOCTUA
Recv: CrashDetect ENABLED!
Recv: FSensor ENABLED (sensor board revision: 0.4 or newer)
Recv: echo:SD init fail
Recv: echo:busy: processing
Printer seems to support the busy protocol, will adjust timeouts and set busy interval accordingly
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: MMU not responding - DISABLED
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing`

leptun, 2022-02-19:
Your issue will be fixed in fw 3.11.0 final release. The following PR will be merged when the time comes: https://github.com/prusa3d/Prusa-Firmware/pull/3390
In the meantime, try sending the following commands to the printer with some other software such as putty:
```
D3 Ax0d28 C1 X00
D3 Ax0d15 C4 X435a5058
PRUSA RESET
```
After the last command is executed (reset), you should be able to connect from octoprint until you do a factory reset in the UI.

leptun, 2022-02-19:
Alternatively, Caribou3d can compile their firmware without PRUSA_SN_SUPPORT defined in the einsy pins definition after #3389 is merged.

jiri-jirus, 2022-02-20:
thanks you. of course I compiled my own firmware earlier with all the prusaSN code commented out.

arekm, 2022-08-27:
@Prusa-Support this can be closed, too.

Prusa-Support, 2022-09-07:
Thanks for the heads up.

This issue is fixed by the feature _fake serial number_ in the _experimental_ menu.
https://help.prusa3d.com/article/experimental-menu-mk3-s-_161213

Michele Moramarco
Prusa Research

closed 2022-09-07 by Prusa-Support

TTTTTAILOR, 2022-11-03:
> Thanks for the heads up.
> 
> This issue is fixed by the feature _fake serial number_ in the _experimental_ menu. https://help.prusa3d.com/article/experimental-menu-mk3-s-_161213
> 
> Michele Moramarco Prusa Research

Thanks,it works well.

PR #3390 (2022-03-04): Fix Octoprint connect because of SN GET on blank einsy boards
Depends on #3389
Fixes #3319 

PFW-1284