prusa3d/Prusa-Firmware #3323: [BUG] PAUSE PRINT feature

rosauro3d, 2021-12-20:
3D printer brand / version + firmware version 
Prusa MK3S, firmware 3.10.0 and 3.10.1

Slicer
PrusaSlicer - Tested with 2.3.1, 2.3.2, 2.3.3

Operating system type + version
Windows 10 Pro, versioni 20H2

Behavior
In the Prusa slicer a PAUSE PRINT was added so that I could embed a nut to a project.

When the print job gets to that point of PAUSING and after adding the hardware I select resume print.
After selecting resume print the screen goes blank for 30 seconds.
The nozzle's temperature continues to drop as the print head continues with the job.
At the same time there is no filament being fed through the nozzle.

I've already reported this with Prusa support back in July 7, 2021.
I've done everything they've instructed me to do. (reflashing firmware, generate new gcode, redownload firmware files, full system refresh, tried slicers 2.3.1, 2.3.2 and 2.3.3, tried Pronterface, tried via SD, generate new gcode for a new project)
I thought the new firmware 3.10.1, that was release a few days ago, would have resolve this issue. Apparently not.
I was told by Prusa support that they will report the issue again and to also report it here as well.

For now I've gone back to firmware 3.9.3 where the PAUSE PRINT last worked.

There is a video I made of the issue that can be viewed here:
http://rosaurophotography.com/misc/videos/prusapause/

gudnimg, 2021-12-20:
I can try to reproduce this on my end (on MKS3+). But I’ve not used this feature before. Do you configure the pause in Prusa Slicer?

I think we should focus on fw v3.10.0

3d-gussner, 2021-12-20:
@rosauro3d have you searched the github for your issue? I think you have the same issue as https://github.com/prusa3d/Prusa-Firmware/issues/3148 describes.

Please read https://github.com/prusa3d/Prusa-Firmware/issues/3148#issuecomment-855201971
and https://github.com/prusa3d/Prusa-Firmware/issues/3148#issuecomment-966463702

Let us know if that helped.

rosauro3d, 2021-12-20:
@gudnimg Yes you configure the pause in Prusa Slicer.  
Check out this video:  https://www.youtube.com/watch?v=CtC5vAK94qA

@3d-gussner Yes I saw that back in July.  It's a temporary solution.  I'm looking for a permanent solution.  I would have thought by 3.10.1 firmware that the issue would have been addressed.

3d-gussner, 2021-12-21:
@rosauro3d There is this working solution, please use it.

I don't know if or when there will be a different solution.
Implementing a host/printer negotiation doesn't sound easy and resource friendly.

The Prusa team is always fighting and improving code to free up resources. There are few feature requests which will consume these limited resources and it is hard to decide which will be included which not.

As M601/M602/M603 with minor modifications in Octoprint is working the chances are high that other features will get a higher priority.

rosauro3d, 2021-12-21:
@3d-gussner 
"There is this working solution, please use it." - No thank you.

Are you saying that Prusa intentionally created a bug so that other feature requests can be implemented?    LOL

That's like saying your vehicle manufacturer disabled your headlights to be able to implement new features.  Then asks you to use a flashlight as a solution...  ...because the development of warp speed has a higher priority....

I will continue to use 3.9.3.

gudnimg, 2021-12-21:
I can clearly see this issue in your video as well https://github.com/prusa3d/Prusa-Firmware/issues/3316

gudnimg, 2021-12-22:
@rosauro3d Could you try and see if the issue appears with this small gcode file. Only takes about 1 minute. The printer will stop after the first layer.
[test_file.zip](https://github.com/prusa3d/Prusa-Firmware/files/7762518/test_file.zip)

I could not reproduce this issue on FW 3.10.1 with PrusaSlicer 2.4.0, I don't think the slicer version should matter.

[image]

rosauro3d, 2021-12-22:
@gudnimg 
I tested your GCODE with FW 3.10.1.  
Here's the [video](http://rosaurophotography.com/misc/videos/prusapause/index.html#gilbert).

gudnimg, 2021-12-23:
I am able to trigger the issue where extruder temperature is not rising afte resuming and LCD still displaying "Print paused" if this lines is hit https://github.com/prusa3d/Prusa-Firmware/blob/0be90dc5d1d8b6b12e063b751c7e16486520ba98/Firmware/ultralcd.cpp#L6390

Though I do not get this 30 second long blank screen, and my printer stays still.

gudnimg, 2021-12-23:
@rosauro3d Are you using Octoprint? If I understand the workaround @3d-gussner correctly it only affects Octoprint?

rosauro3d, 2021-12-23:
@gudnimg 

This is already a known firmware issue with Prusa.  I’ve brought this up with them back in July.  I brought it up to their attention again upon the release of FW 3.101.  The only reason I posted it here is that Prusa support had asked me.  I’m assuming it’s to make everyone else aware and/or to expedite the remedy with the developers.  The temporary fix, in Octoprint (M601 and M602), does work but the issue still remains as it’s clearly evident with the release of FW 3.10.1.

JohnnyDeer, 2022-02-15:
@rosauro3d I got info from our Tech Support that this issue was clarified via email, is that right?

rosauro3d, 2022-02-15:
Hi Johnny

Yes it was clarified that the issue won't be fixed any time soon.  It will
be an ongoing issue.

Regards

Rosauro

agners, 2022-02-27:
Encountered the same issue on a MK3S with FW 3.10.1/PrusaSlicer 2.4.0. Inserted printer pause to pause printing during the night using the "Add pause print (M601)" command in PrusaSlicer. In the morning, pressing the Knob -> `Resume` resumed printing, but tool temperature stayed off (Bed temperature stayed on the whole time, which I think is correct). This is not what I would expect from a pause print command. I am using OctoPrint, but resumed printing using the Knob. I was able to set temperature in OctoPrint, but some part of the layer is now missing. 

Also the display stayed blank since selecting `Resume` (it did not come back even when turning the knob/waiting for more than 30 minutes now). So far, the printer seems to continue fine with the display being blank.

rosauro3d, 2022-04-21:
Issue still the same with firmware 3.11.0

gudnimg, 2022-04-21:
I wonder if the issue is with `lcd_resume_usb_print()`. It would be interesting to change `lcd_resume_usb_print` in the line below to `lcd_resume_print` to see if that fixes the issue. I don't think this will fix the issue where the hotend is turned off, but the display should not go blank.

[image]

I'm thinking this because `lcd_resume_usb_print()` is basically empty. I highly suspect the issue is related to `lcd_draw_update `not being set. If the status menu appears after exactly 30 seconds, then that would confirm it as there is a timer which will expire when there is a 30 second inactivity within the LCD menu and will force the LCD status screen to be re-drawn.

I am unfortunately unable to test this as I can't print over USB.
[image]