prusa3d/Prusa-Firmware #3331: M112 After heating up

sarf2k4, 2021-12-29:
**Printer type** - [e.g. MK3S, MK3, MK2.5S, MK2.5, MK2S, MK2]
MK3S
**Printer firmware version** -  [e.g. 3.8.1, 3.8.1-RC1, ...]
3.10.1-4697

**MMU upgrade** - [e.g. MMU2S, MMU2, MMU1]
MMU2S
**MMU upgrade firmware version** - [e.g. 1.0.6, 1.0.6-RC2, ...]
1.0.6-372

**SD card or USB/Octoprint**
Octoprint Via USB

**Describe the bug**
After heating up the hotend to the target temp, M112 suddenly called out
  
**To Reproduce**
It randomly came at the exact same g-code location, but not 60-100% will be able to reproduce
  
**Expected behavior**
No M112 Issued?
  
**G-code**
Here is my g-code where the M112 often occurred commented below. Yes, this is the start g-code, I only changed S180 instead of the stock settings, the rest are stock g-code from prusa
```
M862.3 P "[printer_model]" ; printer model check
M862.1 P[nozzle_diameter] ; nozzle diameter check
M115 U3.10.0 ; tell printer latest fw version
G90 ; use absolute coordinates
M83 ; extruder relative mode
M104 S180 ; set extruder temp
M140 S[first_layer_bed_temperature] ; set bed temp
Tx
M190 S[first_layer_bed_temperature] ; wait for bed temp
M109 S[first_layer_temperature] ; wait for extruder temp
//My M112 issues always at this point, after the M109 above <===============
G28 W ; home all without mesh bed level
G80 ; mesh bed leveling

;go outside print area
G1 Y-3 F1000
G1 Z0.4 F1000
; select extruder
Tc
; purge line
G1 X55 F2000
G1 Z0.3 F1000
G92 E0
G1 X240 E25 F2200
G1 Y-2 F1000
G1 X55 E25 F1400
G1 Z0.2 F1000
G1 X5 E4 F1000

M221 S{if layer_height<0.075}100{else}95{endif}
G92 E0

; Don't change E values below. Excessive value can damage the printer.
{if print_settings_id=~/.*(DETAIL @MK3|QUALITY @MK3).*/}M907 E430 ; set extruder motor current{endif}
{if print_settings_id=~/.*(SPEED @MK3|DRAFT @MK3).*/}M907 E538 ; set extruder motor current{endif}
```

Following is from the terminal on the Octoprint's web interface on where the M112 occurred.
```
LCD status changed
ok
Full RX Buffer
N72 M109 S230*92
LCD status changed
ok
N73 G28 W*80
N74 G80*34
Error:Line Number is not Last Line Number+1, Last Line: 18
Resend: 19
M112
N75 M112*19
N76 M104 T0 S0*16
N77 M104 T1 S0*16
N78 M104 T2 S0*28
N79 M104 T3 S0*28
N80 M104 T4 S0*29
N81 M140 S0*92
```

**Video**
Don't think I could reproduce this very easily

**Additional Note**
I checked several other 'similar' issues, it might be my thermistor started to acting up due to its age or something, but I could be wrong. As of this writing, previous print job goes without any M112 issued, and I reconnected the octoprint again, start the print job, no M112 issued

JohnnyDeer, 2022-01-03:
Hi @sarf2k4, I would say your assumptions about broken thermistor cable is right. By your description is obvious that M112 appear when axes start moving - cable is bending. Also appearing on same spot in Gcode mean that in that position cable may have broken spot. 
Please try to print this Gcode via USB, insted of M112 should trigger watchdong and say which thermistour is broken. 
Thank you

sarf2k4, 2022-01-03:
Hi @JohnnyDeer, thank you for your input, I quite not understand what you meant by printing the g-code over USB as I am using an octoprint hooked via USB. I am using RasPi 3.

After my first post above, I had few other print jobs like 2-4 jobs, no M112 were triggered at all. 

If I remember correctly, there's a config about printer behavior in octoprint, ".... send M112......" on this, but I'm unsure how I'm going to proceed on this one as M112 might caused by octoprint's side that will trigger the M112 thing. Like maybe, mismatched gcode buffer? I don't know about this part anymore

JohnnyDeer, 2022-01-04:
I meant print without OctoPrint, then mintemps should trigger instead of M122, if it´s caused by broken thermistor. 
Also you can watch LCD before this behavior and see if bed or hotend temp drop, that also will confirm broken thermistor (https://github.com/prusa3d/Prusa-Firmware/issues/3083#issuecomment-804849567).

sarf2k4, 2022-01-05:
@JohnnyDeer, I forgot to mention that the `M112` occurred is not when the printer starts moving, or perhaps it did. I didn't hear any motor moving at all while the `M112` already been issued 'silently'

sarf2k4, 2022-01-24:
I just hit with the `M112` error again, below are from the octoprint's terminal
```
MMU <= 'A'
MMU => 'ok'
mmu_get_response() returning: 1
ok
N18 M113 S2*88
LCD status changed
LCD status changed
ok
Full RX Buffer
N68 M109 S230*87
ok
N69 G28 W*91
N70 G80*38
Error:Line Number is not Last Line Number+1, Last Line: 19
Resend: 20
M112
N71 M112*23
N72 M104 T0 S0*20
N73 M104 T1 S0*20
N74 M104 T2 S0*16
N75 M104 T3 S0*16
N76 M104 T4 S0*20
N77 M140 S0*85
N0 M110 N0*125
start
N0 M110 N0*125
```
In my previous post above, the `G28` & `G80` tests right after heating up didn't trigger any `M112` issues. I do believe it may be due to the RX/TX buffer being full and the firmware couldn't keep up with the data being fed by octoprint. 

Note that I'm also using MMU2S, with all of the newer firmware update on the einsy rambo, would render the internal memory buffer being used up and having much lesser space for the movement planner and such

P.S:
It's just my theory since I'm unable to trigger `M112` issues consistently, even with the series of `G28` & `G80` didn't trigger it either

sarf2k4, 2022-02-13:
I am going to continue dumping terminal log here every time I am getting M112 issues. There are times that I didn't get M112, 
[Terminal M112.txt](https://github.com/prusa3d/Prusa-Firmware/files/8054997/Terminal.M112.txt)

Also, there are others who get the same issues that has been reported over the octoprint forum as well at
https://community.octoprint.org/t/prusa-mk3s-m112-crash-when-start-printing-working-fine-with-sd-card/23526/13

leptun, 2022-02-13:
Nevermind. The ok was supposed to be missing because temperature autoreporting was supposed to be enabled. This brings the question, why was octoprint asking for temperatures manually when the temps were already reported every 2 seconds automatically?

Can you please post the initial connection handshake that happens?
This is what you should see at some point:
```
Changing monitoring state from "Connecting" to "Operational"
Send: N0 M110 N0*125
Recv: ok
Send: N1 M115*39
Recv: FIRMWARE_NAME:Prusa-Firmware 3.10.1 based on Marlin FIRMWARE_URL:https://github.com/prusa3d/Prusa-Firmware PROTOCOL_VERSION:1.0 MACHINE_TYPE:Prusa i3 MK3S EXTRUDER_COUNT:1 UUID:00000000-0000-0000-0000-000000000000
Recv: Cap:AUTOREPORT_TEMP:1
Recv: Cap:AUTOREPORT_FANS:1
Recv: Cap:AUTOREPORT_POSITION:1
Recv: Cap:EXTENDED_M20:1
Recv: ok
Send: M21
Recv: echo:SD init fail
Recv: ok
Send: M155 S2
Recv: ok
```

M155 S2 enables the autoreporting with 2 second interval.
After that, without being prompted, the printer will send this back:
```
Recv: T:23.4 /0.0 B:23.9 /0.0 T0:23.4 /0.0 @:0 B@:0 P:0.0 A:28.9
Recv: T:23.1 /0.0 B:24.5 /0.0 T0:23.1 /0.0 @:0 B@:0 P:0.0 A:29.6
Recv: T:23.4 /0.0 B:24.3 /0.0 T0:23.4 /0.0 @:0 B@:0 P:0.0 A:28.4
Recv: T:23.0 /0.0 B:24.0 /0.0 T0:23.0 /0.0 @:0 B@:0 P:0.0 A:28.9
Recv: T:23.2 /0.0 B:24.1 /0.0 T0:23.2 /0.0 @:0 B@:0 P:0.0 A:29.3
Recv: T:23.2 /0.0 B:24.3 /0.0 T0:23.2 /0.0 @:0 B@:0 P:0.0 A:28.6
Recv: T:23.3 /0.0 B:23.9 /0.0 T0:23.3 /0.0 @:0 B@:0 P:0.0 A:28.9
Recv: T:23.3 /0.0 B:24.0 /0.0 T0:23.3 /0.0 @:0 B@:0 P:0.0 A:29.0
Recv: T:23.2 /0.0 B:23.9 /0.0 T0:23.2 /0.0 @:0 B@:0 P:0.0 A:28.7
Recv: T:23.2 /0.0 B:24.1 /0.0 T0:23.2 /0.0 @:0 B@:0 P:0.0 A:29.6
```

And this is what a preheat should look like. Keep in mind I have no heaters in this setup, so it thermal runaways:
```
Send: N16 M190 S60*105
Recv: LCD status changed
Recv: T:23.1 /215.0 B:24.3 /60.0 T0:23.1 /215.0 @:127 B@:127 P:0.0 A:29.1
Recv: T:23.25 E:0 B:24.1
Recv: T:22.97 E:0 B:23.8
Recv: T:22.9 /215.0 B:24.0 /60.0 T0:22.9 /215.0 @:127 B@:127 P:0.0 A:28.7
Recv: T:23.34 E:0 B:24.0
Recv: T:23.19 E:0 B:24.3
Recv: T:23.3 /215.0 B:23.9 /60.0 T0:23.3 /215.0 @:127 B@:127 P:0.0 A:28.8
Recv: T:23.12 E:0 B:24.1
Recv: T:22.97 E:0 B:23.9
Recv: T:23.1 /215.0 B:23.9 /60.0 T0:23.1 /215.0 @:127 B@:127 P:0.0 A:29.7
Recv: T:23.09 E:0 B:24.1
Recv: T:23.16 E:0 B:24.2
Recv: T:23.0 /215.0 B:23.9 /60.0 T0:23.0 /215.0 @:127 B@:127 P:0.0 A:28.9
Recv: T:23.28 E:0 B:23.9
Recv: T:23.56 E:0 B:24.0
...
Recv: T:23.3 /215.0 B:24.3 /60.0 T0:23.3 /215.0 @:127 B@:127 P:0.0 A:29.0
Recv: T:23.31 E:0 B:24.3
Recv: T:23.31 E:0 B:23.5
Recv: T:23.4 /215.0 B:24.2 /60.0 T0:23.4 /215.0 @:127 B@:127 P:0.0 A:28.9
Recv: T:23.00 E:0 B:23.8
Recv: T:23.06 E:0 B:23.9
Recv: T:23.2 /215.0 B:23.9 /60.0 T0:23.2 /215.0 @:127 B@:127 P:0.0 A:28.9
Recv: T:23.19 E:0 B:23.9
Recv: T:23.22 E:0 B:23.8
Recv: T:23.2 /215.0 B:23.9 /60.0 T0:23.2 /215.0 @:127 B@:127 P:0.0 A:29.5
Recv: LCD status changed
Recv: Error: THERMAL RUNAWAY (PREHEAT HOTEND)
Changing monitoring state from "Printing" to "Cancelling"
Recv: // action:cancel
Cancelling on request of the printer...
Send: N17 M108*28
Recv: ok
Send: N18 M108*19
Recv: Error:Printer stopped due to errors. Fix the error and use M999 to restart. (Temperature is reset. Set it after restarting)
Send: N19 M108*18
Recv: ok
Send: N20 M603*20
Recv: Unknown M code: $17 M108
Recv: ok
Recv: Unknown M code: $18 M108
Changing monitoring state from "Cancelling" to "Operational"
Send: M603
Recv: ok
Send: M603
Recv: Unknown M code: $19 M108
Recv: ok
Recv: // action:cancel
Cancelling on request of the printer...
Recv: ok
Recv: T:23.2 /0.0 B:24.0 /0.0 T0:23.2 /0.0 @:0 B@:0 P:0.0 A:29.0
Recv: ok
Recv: // action:cancel
Cancelling on request of the printer...
Recv: ok
Recv: ok
Recv: T:23.5 /0.0 B:24.3 /0.0 T0:23.5 /0.0 @:0 B@:0 P:0.0 A:29.4
Recv: T:23.1 /0.0 B:24.0 /0.0 T0:23.1 /0.0 @:0 B@:0 P:0.0 A:29.2
Recv: T:23.3 /0.0 B:24.3 /0.0 T0:23.3 /0.0 @:0 B@:0 P:0.0 A:29.1
Recv: T:23.2 /0.0 B:24.1 /0.0 T0:23.2 /0.0 @:0 B@:0 P:0.0 A:29.6
Recv: T:23.1 /0.0 B:24.1 /0.0 T0:23.1 /0.0 @:0 B@:0 P:0.0 A:29.3
```

Notice how in my case octoprint never sends any M105 to get the temperatures. 

Please also post this section of the octoprint settings:
[image]

sarf2k4, 2022-02-13:
Here you go, there are 2 connection establishment there just to be sure everything is in this file
[terminal.txt](https://github.com/prusa3d/Prusa-Firmware/files/8056698/terminal.txt)

Following is the system info copied directly from octoprint's setting
```
browser.user_agent: Mozilla/5.0 (Windows NT 10.0; Win64; x64; rv:97.0) Gecko/20100101 Firefox/97.0
connectivity.connection_check: 1.1.1.1:53
connectivity.connection_ok: true
connectivity.enabled: true
connectivity.online: true
connectivity.resolution_check: octoprint.org
connectivity.resolution_ok: true
env.hardware.cores: 4
env.hardware.freq: 1200
env.hardware.ram: 914010112
env.os.bits: 32
env.os.id: linux
env.os.platform: linux
env.plugins.pi_support.model: Raspberry Pi 3 Model B Rev 1.2
env.plugins.pi_support.octopi_version: 0.17.0
env.plugins.pi_support.throttle_state: 0x0
env.python.pip: 20.3.3
env.python.version: 3.7.3
env.python.virtualenv: true
octoprint.safe_mode: false
octoprint.version: 1.7.3
printer.firmware: Prusa-Firmware 3.10.1 based on Marlin
systeminfo.generator: systemapi
```

Here is the screenshot of the settings you requested
[image]
[image]
[image]
[image]

leptun, 2022-02-13:
We are using the same octoprint version, my python is newer but that shouldn't matter.

The initial connection looks mostly correct, except this:
```
Recv: Command not found!
```
That is the 32u2 chip (the usb to serial) that complains. Not sure how it got into that state. It might help if you tick "wait for start".
Were Tx and Tc really needed in "Long running commands"? They are supposed to provide feedback about the busy state and I think they aren't needed there. What happens if you remove those?

Can you please also post logs with the beginning of the print? (starting from before hitting print till M112). Btw, I think there's a serial log setting in octoprint which also includes timestamps and is not limited in length. It might be helpful to enable that for debugging.

sarf2k4, 2022-02-14:
> We are using the same octoprint version, my python is newer but that shouldn't matter.
>
> The initial connection looks mostly correct, except this:
>
> ```
> Recv: Command not found!
> ```
>
> That is the 32u2 [...]

I added `Tc/Tx` into the long running commands because in my own opinion, the `Tx` is only load the selected filament to the extruder gear, until the sensor has been triggered, and extrusion operation can be done. If I remember correctly, without `Tc` included, the mmu2/mainboard are in a state of hanging or something. Perhaps I'm wrong that `Tc` must be included, I don't know on this part

Right now, I've removed `Tx/Tc` from the long running commands, enabled serial logging for debugging. Also, it will be a while for me to get the mysterious `M112` error as I very rarely do any print jobs. From my observation, I would get this `M112` on the first print of that day like the previous print were 2-4 days ago, then `M112` error would come back again. So it's going to be a hit and miss to get the log file after this

leptun, 2022-02-14:
The point of the "long running commands" is so that octoprint knows that the commands do not send back a response. But these commands MUST send back busy messages. Actually, (I think) all commands in PrusaFirmware send busy messages. I'm not sure if octoprint does something funny when a command in that list is sent. Can you make a backup of your octoprint instance and reset all settings to default? I'd like to rule out the possibility of something like that causing issues.

The einsy has (and always had) 2 microcontrollers: An atmega2560 (the main microcontroller) and an atmega32u2 that handles the USB to UART link. The 32u2 also has a subset of commands (see https://cfl.prusa3d.com/display/PI3M3/UsbSerial+commands) since the serial number of the board is stored in the eeprom of the 32u2 by default. The language IC that you say is a SPI attached flash memory (256KB). This is the chip that holds the langauges when you upload a firmware. When you select a language in the menu, the language data for that specific langauge gets copied to the internal flash of the atmega2560.

sarf2k4, 2022-02-14:
> The point of the "long running commands" is so that octoprint knows that the commands do not send back a response. But these commands MUST send back busy messages. Actually, (I think) all commands [...]

Should I post a logged serial here with current settings intact or reset all settings to default before trying to get hit with the `M112` error later

leptun, 2022-02-14:
Keep the settings like they are now until you get the first M112. Then after we have a good serial log we can try a fresh config

sarf2k4, 2022-02-14:
> Keep the settings like they are now until you get the first M112. Then after we have a good serial log we can try a fresh config

Okay, I've already enabled the serial logging in the settings, what's left is to wait for the event to happen. I assume I can just directly download it in the log section right?

sarf2k4, 2022-02-19:
Good news, as what I've expected, `M112` has struck again. Before this, I've updated a plugin "UI Customizer"
[octoprint-systeminfo-20220219175652.zip](https://github.com/prusa3d/Prusa-Firmware/files/8101891/octoprint-systeminfo-20220219175652.zip)

leptun, 2022-02-19:
We've found the issue in the meantime. Octoprint was treating `Tx` as non-gcode and wasn't waiting for an "ok" response from that command. Octoprint 1.8.0 will have a patch for this and in the firmware we've already added a MMU2 capability line (in the MK3 branch) which should inform the host software of the `T` gcode extensions (Tx, Tc, T?). The change in the firmware will come in the 3.12 firmware release. See also https://github.com/OctoPrint/OctoPrint/issues/4423, https://github.com/OctoPrint/OctoPrint/issues/4422, https://github.com/prusa3d/Prusa-Firmware/pull/3180#issuecomment-1041427206 and https://github.com/prusa3d/Prusa-Firmware/issues/3411

sarf2k4, 2022-02-19:
> We've found the issue in the meantime. Octoprint was treating `Tx` as non-gcode and wasn't waiting for an "ok" response from that command. Octoprint 1.8.0 will have a patch for this and in the firmware we've already added a MMU2 capability line (in the MK3 branch) which should inform the host software of the `T` gcode extensions (Tx, Tc, T?). The change in the firmware will come in the 3.12 firmware release. See also [OctoPrint/OctoPrint#4423](https://github.com/OctoPrint/OctoPrint/issues/4423), [OctoPrint/OctoPrint#4422](https://github.com/OctoPrint/OctoPrint/issues/4422), [#3180 (comment)](https://github.com/prusa3d/Prusa-Firmware/pull/3180#issuecomment-1041427206) and #3411

Finally... thank you for recognizing this is an issue exclusively for mmu2s. I also suspected it was actually the `Tx` command was the problem, after suspecting it might as well be full rx buffer. While some suggested that my thermistor cable were broken on somewhere but the printer didn't move when the `M112` were triggered.

I hope we all will get this fix very soon, this issue has been here for quite a while now, like a year or more, even for other users.

Once again, thank you for having your attention to this issues that will lead this to be fixed. Have a good day

knobik, 2022-02-21:
@sarf2k4 in the mean time, if you are using octoprint, this should fix it as a "dirty" hack https://github.com/OctoPrint/OctoPrint/issues/4422#issuecomment-1041537394

sarf2k4, 2022-03-05:
> @sarf2k4 in the mean time, if you are using octoprint, this should fix it as a "dirty" hack [OctoPrint/OctoPrint#4422 (comment)](https://github.com/OctoPrint/OctoPrint/issues/4422#issuecomment-1041537394)

I somewhat would like to avoid any non-conventional modifications where i may forgot i did that changes. I think a cumbersome yet non-intrusive workaround is to preheat the heaters first so octoprint won't timeout, or increase the timeout instead

closed 2023-11-14 by github-actions[bot]