prusa3d/Prusa-Firmware #3332: [QUESTION] Is there a particular reason that 1193 (pwm fan noise fix) is not implemented yet on the release firmware?

arminAnderson, 2021-12-31:
Stock MK3S+, on 3.10.0

I've been trying to diagnose my very loud fan noise now that I'm working with petg, and came across [this conversation](https://github.com/prusa3d/Prusa-Firmware/pull/1193), which is my exact problem. All my reading on the repo about this topic gives me the impression that this solution works well, however I looked at the current version of the files and it seems the changes are still not implemented. I understand there is a downside with the fan speed readouts, but is there a specific reason this is not implemented as a option on the lcd or something (or maybe it is and I'm just not seeing it)? This seems like a very common problem, and I'm seeing Prusa sending replacement fans for this specific noise issue; I feel like I'm missing something on why this is not available in the release firmware as a user option.

The last on this issue was back in [January 2021](https://github.com/prusa3d/Prusa-Firmware/issues/2943), so I was wondering is there an update on this?

agravelot, 2022-10-09:
Cannot agree more on this one. I think silent mode (or a more advanced one) will be totally appropriate.

I'm thinking to get a refund because it's driving me nuts.

Prusa-Support, 2022-10-18:
Thank you for reaching us.

It seems you are referring to an incomplete issue (clarified and closed by the author) and a Pull Request which wasn't merged for some reason (maybe not safe enough) but closed.
More about the latter, it was actually closed about the time that FW [3.5.2-RC1](https://github.com/prusa3d/Prusa-Firmware/releases/tag/v3.5.2-RC1) was published. Our developers worked for quieter fan operation on that firmware release, which apparently satisfied the community. 

There could be various reasons for the fan to be noisier than usual.
You can get in touch with our official support for further help.
https://help.prusa3d.com/article/customer-support_2287

I hope this answers your question and clarifies any doubts about the matter. If so, please close this issue.

Michele Moramarco
Prusa Research

Prusa-Support, 2022-11-08:
The problem mentioned in this issue is unconfirmed. Also, the questions concern aspects that have been supposedly covered in past issues and firmware releases.
In case of concerns about a noisy fan, please contact the official Customer Support providing a video for evaluation.
https://help.prusa3d.com/article/customer-support_2287

I hope you don't mind me closing this issue now.

Michele Moramarco
Prusa Research

closed 2022-11-08 by Prusa-Support

agravelot, 2022-11-09:
The issue seems quite well explained on linked comment by author : https://github.com/prusa3d/Prusa-Firmware/pull/1193#issuecomment-430587423

Prusa-Support, 2022-11-23:
Yeah. What I'm trying to say is that the comment you are referring to is older than the firmware that supposedly addressed the concern: apparently, sometime later, our developers came out with a workaround via firmware upgrade, that solves the issue of noise.
Please check the release notes of the firmware linked above.

It is rather safe to say that comments before February 2019, or generally referred to older firmware releases (before 3.5.2-RC1) are obsolete at this point.

In case of specifically noisy fans, please reach our Support.

Michele Moramarco
Prusa Research