prusa3d/Prusa-Firmware #3333: [QUESTION] What is the correct bed temp when sensor is disconnected?

marcopiraccini, 2022-01-01:
I am using a MK2.5S with latest FW (`3.10.1-4697`).
The problem I have is that the bed temp is now always `14`, but it's `14` **even if I disconnect the bed temperature sensor from the board!**
So my question is: is this is the current firmware behavior? 
(If not, I guess my mini-rambo board is damaged :( ).

leptun, 2022-01-01:
See this comment. It’s a design flaw in the electronics that prevents
reading temperatures under 14C on the miniRambo. The EinsyRambo is not
affected by this.
https://github.com/prusa3d/Prusa-Firmware/pull/1855#issuecomment-517601792

marcopiraccini, 2022-01-01:
Oh interesting! So it seems that the board behavior is "correct". Closing this, tnx.

closed 2022-01-01 by marcopiraccini