prusa3d/Prusa-Firmware #3335: [QUESTION]Disable MMU retraction after FINDA senses filament is out

IllicLanthresh, 2022-01-02:
Hi, I have a stock MK3S with an MMU2S and I've been trying with no luck to stop my MMU from retracting the last couple cms of filament when FINDA detects a filament runout.

I want to prevent having to throw away those small filament sections that the MMU refuses to use. Is there a way to make the print stop only once the filament sensor on the print head detects it and then continue with the usual spool join procedure?

How else can I use my filament spools until the last drop?

Thanks.

vintagepc, 2022-01-02:
One potential pitfall with this is that once the extruder senses a runout the filament is past the gears and can no longer be removed because the motor has no grip on it. So if you end up with a jam due to a bend on the end of the roll, or some other feeding issue caused by the two ends not feeding properly you now have to do a complete extruder/hotend disassembly to fix the issue...

leptun, 2022-01-02:
Yes the retractions might be an issue while the small stub is being eaten by the hotend. An alternative would be to push the entire stub out during the automatic tool change. This way is a bit more wasteful than the ideal case, but avoids the retraction issues. I think the only other problem that this approach might have is that the user is needed at the end of the print to remove the strand manually if the tip of the filament that is currently loaded when the print finishes is already past the finda and can't be grabbed by the mmu anymore.
However, I actually quite like this single material MMU print with spooljoin where the entire idea is to switch from one spool to another with the minimum amount of waste. I actually see this working much betten than unloading a filament to the mmu once the tip is past the finda (quite often fails due to the shape of the filament end).

@DRracer fyi

IllicLanthresh, 2022-01-03:
> One potential pitfall with this is that once the extruder senses a runout the filament is past the gears and can no longer be removed because the motor has no grip on it. So if you end up with a jam due to a bend on the end of the roll, or some other feeding issue caused by the two ends not feeding properly you now have to do a complete extruder/hotend disassembly to fix the issue... and lose the print in the process.

I am aware that it could result in filament getting stuck inside the ptfe tube between the MMU and the print head, but I do not quite understand how that could result in having to disassemble the print head, at worst you lost a ptfe tube because you are not able to remove the stuck filament inside. 

Correct me if I'm wrong but I remember my MK3 printing until the last drop without the MMU upgrade, why would now that be a problem after just adding the MMU?

Also, as far as I know, prusa already puts a disclaimer in the [documentation](https://help.prusa3d.com/en/article/spooljoin-mmu2s_134252) requiring you to cut the end of the filament to remove the bent tip so it doesn't get jammed while unloading on the MMU.

> Yes the retractions might be an issue while the small stub is being eaten by the hotend. An alternative would be to push the entire stub out during the automatic tool change. This way is a bit more [...]

Another possible option is to make the user aware of the situation and if in any case he needs to remove that filament strand, give him the option(along with pushing the remaining filament) to remove the ptfe tube end from the print head and do a manual filament unload.

vintagepc, 2022-01-03:
I'm not talking about the feed to the print head from the MMU, I'm talking about the PTFE inside the hot-end. 

Remember, the standard MK3 has the filament sensor *BEFORE* the gears, so the extruder motor still has grip on it when the runout is sensed. 

With the MMU chimney the sensor *is* the extruder gears, so if that reads zero you have no guaranteed way to pull the filament out of the hot zone anymore and the only thing you can do is continue feeding more filament after it. 

If the new piece you're following it with doesn't bump up perfectly and provide a solid push, you can have feed issues - I've definitely had it happen myself when manually "chasing" the end of a spool with a fresh one on a standard machine - the new piece didn't quite follow right and either bound up on the edge of the PTFE and jammed or slid off to the side rather than going where it should.

Not saying it's a guaranteed issue, just that there's a non zero risk of such problems I'm not sure is worth saving ~30cm of filament. I just save those scraps and use them for manual pre-purging from different material types or high contrast colour changes.

leptun, 2022-01-03:
@vintagepc On my MMU2S extruder, even though the filament sensor senses the gears only, the filament runout unloading works even when using the gears sensor only (no mmu connected). However, I think this is due to perfect chimney positioning, something that can't really be expected from every printer.
I've also done the on the fly filament feeding before without the multimaterial to minimize waste and I've never really encountered the issue you are describing. It sounds possible, but honestly improbable. Also, isn't the hotend ptfe chamfered at the end closer to the gears?

vintagepc, 2022-01-03:
This was a while ago, didn't the 3S have an extruder redesign that constrained the path a bit better? Certainly could be less of an issue now.

IllicLanthresh, 2022-01-03:
@vintagepc apologies for that, I didn't understand what you meant. 

Even though the risk exists, I personally think it's quite improbable if you warn the user about it and prompt for confirmation on the end tip being correctly shaped when activating such behavior. Also, this behavior would be non-standard and opted by the user on purpose minimizing the chances even more.

IllicLanthresh, 2022-01-03:
btw, should I swap the issue to a feature request? I didn't know this was something not originally planned and I was missing something in the settings or something similar.

closed 2023-11-14 by github-actions[bot]