prusa3d/Prusa-Firmware #3352: [FEATURE REQUEST] Allow cancellation of Live-Z WITHOUT saving adjustment

alexwhittemore, 2022-01-20:
I think this might be a duplicate of https://github.com/prusa3d/Prusa-Firmware/issues/1968, but I honestly can't tell.

If it makes sense, enter what type of printer or upgrade the feature request applies to.
**Printer type** - Mk2.5, but I think all of the Mk<=3 series
**MMU Upgrade** - N/A

I'm a power-user of "press and hold knob to change Z height." I.e. I very regularly finish a print, then press and hold to move the nozzle away. Or more often, cancel a failed print, then press and hold to move the nozzle away.

Helpfully, a few firmware updates ago introduced the ability to press-and-hold the knob to get to the Live Z adjustment while in-print. That's cool, since changing the Z height in a running print makes no sense.

Unhelpfully, the definition of "in a print" is VERY fuzzy, and I literally-every-time I cancel a print try to jog the extruder up, only to accidentally enter the live Z menu and mess it all up.

Since there's no option to cancel a live Z adjustment, my only recourse is to reset the printer at this point.
  
**Describe the solution you'd like**
In the Adjusting Live Z modal, I'd LOVE a "press and hold to cancel." I.e. tapping the knob commits the new live Z, where long-pressing cancels out without saving the value.

Panayiotis-git, 2022-01-22:
The long-press functionality as implemented at the moment is inconsistent. Look at the https://github.com/prusa3d/Prusa-Firmware/pull/3014#issuecomment-776993759 for more details. I believe that this event requires a clear definition, so a user is always sure what to expect from a long-press.

gudnimg, 2022-01-23:
Just to highlight the inconsistencies mentioned by @Panayiotis-git below is the menu when the printer is not printing anything. Obviously not all the menus.

| Menu                 |  Long-Press | What does it do? |
| ----------------- | -------------------- | ---------------  |
| Status      | X | Open move Z-axis menu   |
| Preheat | X | Open move Z-axis menu   |
| Print from SD card | X | Open move Z-axis menu   |
| AutoLoad filament| X | Exit menu   |
| Unload filament|  |  |
| Settings  | X | Open move Z-axis menu   |
| Calibration | X | Exit menu    |
| Statistics | X | Exit menu    |
| Fail stats| X | Exit menu    |
| Support | X | Open move Z-axis menu   |
| Settings->Temperature  | X | Open move Z-axis menu   |
| Settings->Move axis  | X | Exit menu   |
| Settings->Crash det. | X | Exit menu   |
| Settings->HW Setup  | X | Nothing  |
| Settings->Mesh Bed Leveling  | X | Exit menu   |
| Settings->Lin. correction  | X | Exit menu   |
| Settings->Live Adjust Z  | X | Open move Z-axis menu   |
| Settings->Select Language  | X | Exit menu  |
| Settings->Brightness  | X | Exit menu  |

leptun, 2022-01-23:
@gudnimg I've updated the HW setup "What it does" column

alexwhittemore, 2022-01-23:
TIL that you could move the Z axis while you were already in the middle of live adjust Z. Is that even real?

closed 2023-11-14 by github-actions[bot]