prusa3d/Prusa-Firmware #3359: [BUG]<Why M601 is needed sometimes for M220 to work?>

laptop-laptop, 2022-01-22:
**Printer type** - [e.g. MK3S, MK3, MK2.5S, MK2.5, MK2S, MK2]
i3MK3S+

**Printer firmware version** -  [e.g. 3.8.1, 3.8.1-RC1, ...]
3.10.1 and previous version

**SD card or USB/Octoprint**
SD card

**Describe the bug**
  
Sometimes having M220 S30 in the gcode does not change the overall print speed. Actually from the LCD I saw it changed to 30% but then immediately went back to 100%. Sometimes having M220 alone without M601 can change the speed but sometimes I need to issue M601 after M220 for the speed to change. How come? Under what situation can I change the speed with only M220?

gudnimg, 2022-01-22:
I am unable to reproduce this. How frequently does this issue occur?

Could you share the relevant gcode file?

JohnnyDeer, 2022-01-24:
Hi @laptop-laptop, thank you for reporting. 
As @gudnimg wrote, can you please upload Gcode, for testing?

laptop-laptop, 2022-01-26:
Thank you for the replies. Basically I just replaced M600 in gcode generated by Prusa Slicer 2.4.0 by the following:

M117 Change Filament
M600
M117 Change Filament
M220 S30  
M601

By default, the overall print speed is 100%. I noticed that if I have M220 Sx, where x is percentage below 100, I will need to execute M601 for the printer to change to x% speed. Otherwise, it just showed x% in the LCD and then immediately changed back to 100%. On the other hand, if x is 100, then I  don not need to issue M601 after that for it to resume the speed to 100%.
For example, I changed the speed in the middle of the print from 30% to 100%. All I need to do is to issue M220 S100 without M601 after it. However, if it is printing at 100% and I need to reduce the speed to 30%, I need to have M601 after M220 S30 in other for the speed to change to 30%. Otherwise, it just displays 30% on the LCD screen and quickly changed back to 100%.

BTW, whatever M117 just showed up on the LCD screen for less than a second. Can I make it stays on the screen longer?

closed 2023-11-15 by github-actions[bot]