prusa3d/Prusa-Firmware #3361: [BUG] Settings/Calibrate/Calibrate Z should not have a "right arrow"

gudnimg, 2022-01-23:
**Printer type** - [e.g. MK3S, MK3, MK2.5S, MK2.5, MK2S, MK2]
**Printer firmware version** -  3.10.1

**MMU upgrade** - N/A
**MMU upgrade firmware version** -  N/A

**SD card or USB/Octoprint**
 N/A

**Describe the bug**
I've interpreted menus that **don't** have the right arrow to mean that if you select that menu then there is immediate action from the printer i.e. the printer will do something. And when a right arrow is present then that means there is an additional submenu below. This is not the case with Calibrate Z.
  
[image]

**To Reproduce**
N/A
  
**Expected behavior**
I would expect there to not be a right arrow next to "Calibrate Z" just like "Calibrate XYZ"

leptun, 2022-01-23:
Noticed this issue as well.
Not sure why MENU_ITEM_SUBMENU_P was used in place of MENU_ITEM_FUNCTION_P. I think that is a bug or it was done this way to hide some other issue. This should be investigated.
If you see any other inconsistencies like this, please add them to the issue report.

Panayiotis-git, 2022-01-23:
Is there a reason that the Move X, Move Y and Move Z follows the exact same logic? They all are a single field form that require the use of the rotation wheel to change a value.

leptun, 2022-01-27:
The "Move" menu items are in fact submenus. Even though it doesn't look like it, they are.
Think of it like this: Does clicking the item change the screen to something else entirely? Then it's a submenu. Otherwise, the view will stay right there (like with a toggle or when pressing Auto Home), so it's just a menu item (not a submenu), hence why it should have no arrow in that case.

I think all menu items should have standardized rules. The current situation is suboptimal, especially considering that the older menu items seem not to confirm to how menus look now.

Here is another example of an inconsistency:
In some menus the "back -^" menu text field actually says the name of the menu above, not the generic "Back". Not all menus do this and some have nasty hacks so that the correct back message is displayed. If there are going to be any changes to the UI, I think it might be worth making everything consistent (so a standard must be decided on) and some of these hacks need to be toned down a bit since they make the UI code really ugly and don't really add anything to the look of the UI.

Panayiotis-git, 2022-01-27:
To be honest, I misunderstood the case. I though we were talking about the Live z-adjust... :blush:

gudnimg, 2022-01-29:
Then there is also menu items like this `MENU_ITEM_GCODE_P(_T(MSG_AUTO_HOME), PSTR("G28 W"));` which use `MENU_ITEM_GCODE_P`. The "Calibrate Z" is use a gcode command but by using `MENU_ITEM_FUNCTION_P` we get the ability to return the printer to the status screen after clicking it. IMO these two menus don't need to be different in behaviour.

leptun, 2022-02-19:
I tagged this for 3.12. We should design the standard, document it somewhere (possibly at the beginning of ultralcd.cpp) and then apply it to all menus, submenus and menu items. However, this task is low priority for now

3d-gussner, 2023-12-19:
@gudnimg @leptun Created PR #4543 to address this. Can we close this one?

3d-gussner, 2024-04-03:
Closing as PR https://github.com/prusa3d/Prusa-Firmware/pull/4650 fixes it

PR #4650 (2024-04-04): LCD_Menu_cleanup
New version of https://github.com/prusa3d/Prusa-Firmware/pull/4543 with minor changes

Change to `MENU_ITEM_FUNCTION_P`

- [X] Calibrate Z
- [X] Thermal Model cal.

Fixes #3361

This PR will break after https://github.com/prusa3d/Prusa-Firmware/pull/4565 but not a big issue.
commit b3a410ab (2024-04-04): Fix #3361