prusa3d/Prusa-Firmware #3363: [FEATURE REQUEST] MMU2S disable/bypass option through FIRMWARE

vladturdean, 2022-01-27:
Since I can't print as fast with the MMU2s, I would really like a option where you can disable/enable the MMU2s without removing the wiring whilst printing directly, as usual.

krepsy3, 2022-02-11:
Seconded. I need this very much. When printing with large diameter nozzles, the filament tip is getting stuck CONSTANTLY in the extruder area. Its so tiring to be constantly removing the door and manually removing bits of wasted filament... For single mode prints, I see this as an absolute NECESSITY!

leptun, 2022-02-19:
This feature is coming to fw 3.12 along with the new mmu2 firmware. See https://github.com/prusa3d/Prusa-Firmware/issues/1876#issuecomment-1045965924

gudnimg, 2023-07-31:
Closing as this is now resolved with FW 3.13.0 and MMU FW 3.0.0 :)

closed 2023-07-31 by gudnimg