prusa3d/Prusa-Firmware #3366: [BUG]when starting a print without filament, it will ask to unload filament before offering to load

evils, 2022-01-30:
this appears to be a duplicate of #2501

**Printer type** - MK3S+
**Printer firmware version** - 3.10.1

**SD card or USB/Octoprint**
  it definitely occurs with Octoprint, presumably also from SD card

**Describe the bug**
  M600 (presumably) is called when the filament sensor reports no filament after bed leveling
  (i prefer holding off on loading filament until after bed leveling to avoid oozing)
  this calls for a removal of the old filament before the new filament can be inserted
  (**several beeps and button presses are involved before one can load filament**)
  despite the firmware knowing from the sensor that there is no filament
  (if filament was detected, it wouldn't call for M600)
  
**To Reproduce**
  1. unload filament
  2. start a print
  3. get asked to remove filament that isn't there
  optionally:
  4. load filament
  5. start print
  6. note that it does not go through a filament change routine because filament was detected
  
**Expected behavior**
  the same behaviour one gets from the "auto load filament" main menu option (M701?)
  (it already knows there is no filament, it should just beep once, give the user enough Z clearance to inspect the nozzle, grab the filament when insertion is detected, and ask if it's the right colour)

gudnimg, 2022-02-12:
I can confirm the printer asks me to unload when there is no filament in the printer.

gudnimg, 2022-02-12:
https://youtu.be/pUrLCVDf3UM Heres a video of it. Skip to around 2:07 to see the Unload message when there is no filament in the printer. 

I am using the current MK3 branch in this video.

leptun, 2022-02-12:
I've seen this as well. I'm not sure what would be the best approach to solve this. The runout happens when the extruder moves and the filament sensor says there isn't any filament. That's why it thinks it's a runout so late after starting the print. Other than preventing the start of a print without filament (which I don't like), I'm not sure if there is any easy way arond this problem.

evils, 2022-02-13:
i hadn't considered this could be runout detection

if the firmware could keep track of the filament, the need for an unload could be determined
if it loaded filament at some point, and it was never unloaded, there likely is material in the extruder that could become a stub

here's some pseudocode
```
bool stub = false;

fn main() {
  ...
  extrude();
};

fn extrude() {
  if (!filament) load();
  if (filament) stub = true;
  ...
};

fn load() {
  if (stub) unload();
  extrude();
};

fn unload() {
  ...
  stub = false;
};
```
(main -> extrude -> load -> unload)

maybe such a stub bit could be persisted between power cycles without incurring too much wear on the non-volatile storage?

Panayiotis-git, 2022-02-13:
A filament run-out condition can only happen from a move extruder command. Would it be enough to check the filament sensor at the start of each movement to get the baseline value?

If filament not present:
- On start of extruder move, trigger load filament
- During extruder move, trigger run-out condition

evils, 2022-02-14:
> Would it be enough to check the filament sensor at the start of each movement to get the baseline value?
>     * On start of extruder move, trigger load filament

that's not enough, there may be a "stub" left in the extruder from a previous unresolved runout
(remainder of filament below the filament sensor, but still in the gears)
the printer needs to keep track of that, and unload it before loading new filament

i think the current "always unload before loading" behaviour is because there's a high chance of errors/jams when loading new filament on top of a stub
you could push filament against the stub, triggering the filament sensor, but then not have the new filament caught by the gears
or end up pushing the stub down and jamming the hotend because it's not connected to the new filament
(imagine a thinner end on both the new filament and the stub, it could wedge against the walls instead of pushing down)

both cases end up with a piece of filament in the hotend with no way to pull it out

hagan66, 2023-08-11:
I have this as well, MK2.5S with MMU2S. If I lighten the idler tension it goes away, for example when printing TPU.

closed 2023-11-15 by github-actions[bot]