prusa3d/Prusa-Firmware #3384: [BUG]<MK3S - 3.10.1-4697 "No SD Card" menu item after upgrade to Dec 16 2021 firmware>

560tech, 2022-02-09:
SD card is detected on booting or insertion but shows "Main" and no content for an SD card formatted to FAT32
and tested working with other printers and windows laptop.

This user has same issue:
https://forum.prusaprinters.org/forum/original-prusa-i3-mk3s-mk3-hardware-firmware-and-software-help/fw-update-sd-card-not-being-read-properly-printer-crashing/

1/ Swapped IDC cables 1 and 2 			- SD card contents not displayed.
2/ Swapped IDC cables for new set 		- SD card contents not displayed.
3/ Replaced LCD module with new			- SD card contents not displayed.
4/ Regressed 3 firmware versions		        - SD card contents not displayed.
5/ Replaced Einsy/Rambo board with new 	- SD card contents displayed.

Change in MK3S firmware 3.10.1-4697 that is not reset by regressing firmware versions.

Einsy Rambo board revision 1.2b
QR/Barcode: 036101609

I see this line in the changelog:

https://github.com/prusa3d/Prusa-Firmware/releases/tag/v3.10.1

"Message on boot while inserting SD card fixed
When the user inserted the SD card during the booting process, the "Sorting files" message was presented twice on the display later followed by the correct "Print from SD menu" message. The issue has been reported on GitHub issue #3135 and clearly reproduced.
We have fixed the SD card handling while booting scenario, which corrected the double "Sorting files" message. Now the system works correctly.

leptun, 2022-02-09:
This really sounds like a hardware issue, but a software bug is not out of the question.

Can you please connect to the printer via usb with some terminal software so we can see the messages sent while it tries to read the sd card? You can use Putty or Pronterface or Octoprint or whatever you want

560tech, 2022-02-09:
The SD card worked every day even while I was servicing the extruder.
Only after successfully flashing the upgrade did it stop reading the
contents.

https://forum.prusaprinters.org/forum/original-prusa-i3-mk3s-mk3-hardware-firmware-and-software-help/fw-update-sd-card-not-being-read-properly-printer-crashing/

560tech, 2022-02-09:
That's another user losing SD card functionality after flashing to 3.10.1

560tech, 2022-02-09:
> will try putty and/or pronterface, never used pronterface

I now have to remove the spare board from the 5th printer i was going to
build this week that is now in the printer in question to check the lcd
module and swap back in the original board. more time.

leptun, 2022-02-09:
You can also try to swap lcd assemblies between printers and see if the electronics is at fault or the lcd module/cables.
If it's the lcd, then it's a faulty unit. If it's the einsy, it could be a bad board or a software error.