prusa3d/Prusa-Firmware #3385: [BUG] MK2.5S (probably all others) bed max temp is 115 degreeds celsius during print

galagithub, 2022-02-10:
1) Start a print
2) Go to Tune-Bed in LCD menu
3) Attempt to set it above 115 degrees Celsius
4) Get mad and throw the printer out the window because you just decreased the bed temp from 120 degrees during a print !

galagithub, 2022-02-10:
Firmware 3.10.1-4697
Date: Dec 16 2021

Official firmware !

galagithub, 2022-02-10:
```
diff --git a/Firmware/ultralcd.cpp b/Firmware/ultralcd.cpp
index daac8d71..3943aa4a 100755
--- a/Firmware/ultralcd.cpp
+++ b/Firmware/ultralcd.cpp
@@ -6881,8 +6881,8 @@ static void lcd_tune_menu()
 	MENU_ITEM_BACK_P(_T(MSG_MAIN)); //1
 	MENU_ITEM_EDIT_int3_P(_i("Speed"), &feedmultiply, 10, 999);//2////MSG_SPEED c=15
 
-	MENU_ITEM_EDIT_int3_P(_T(MSG_NOZZLE), &target_temperature[0], 0, HEATER_0_MAXTEMP - 10);//3
-	MENU_ITEM_EDIT_int3_P(_T(MSG_BED), &target_temperature_bed, 0, BED_MAXTEMP - 10);//4
+	MENU_ITEM_EDIT_int3_P(_T(MSG_NOZZLE), &target_temperature[0], 0, HEATER_0_MAXTEMP);//3
+	MENU_ITEM_EDIT_int3_P(_T(MSG_BED), &target_temperature_bed, 0, BED_MAXTEMP);//4
 
 	MENU_ITEM_EDIT_int3_P(_T(MSG_FAN_SPEED), &fanSpeed, 0, 255);//5
 	MENU_ITEM_EDIT_int3_P(_i("Flow"), &extrudemultiply, 10, 999);//6////MSG_FLOW c=15
```
And this fixes it!

leptun, 2022-02-10:
I'm sorry but I can't find the offending code. I don't remember ever changing to maxtemp instead of maxtemp - 10

galagithub, 2022-02-10:
> I'm sorry but I can't find the offending code. I don't remember ever changing to maxtemp instead of maxtemp - 10
try branch MK3_3.10.1

cauta si vei gasi, e din Marlin luat.

galagithub, 2022-02-10:
> ```
> 	MENU_ITEM_EDIT_int3_P(_T(MSG_NOZZLE), &target_temperature[0], 0, HEATER_0_MAXTEMP - 10);//3
> 	MENU_ITEM_EDIT_int3_P(_T(MSG_BED), &target_temperature_bed, 0, BED_MAXTEMP - 10);//4
> ```
> 
> This is correct. If you set the target temperature to exactly maxtemp, the maxtemp error will trigger since the temperature will most likely get a bit over that maxtemp threshold. This is why it is limited to 10deg below maxtemp

This is incorrect since the specs say you can use 125 degrees celsius on the bed.
Secondly, hardcoding crap in the code makes if for an unusual and buggy behavior!

Your own PID setting routine subtracts 3 degrees from MAXTEMP, so which is the correct behavior?

If you really want to take care of the thermal runaway error, which is likely you will trigger if you set the bed too MAX_TEMP, then add an overshoot parameter, 5 degrees should suffice.

vintagepc, 2022-02-10:
Just an FYI - we are mostly all just volunteer/external collaborators here, not official Prusa employees. And while I get you're upset, your message does come across as a bit rude and demanding - that code appears to have been in place for a very long time by someone not in this thread, and the fix is (potentially) not as trivial as you propose because there is some further interplay with additional correction factors that would need testing to be approved and merged. My thoughts here is that change was made for a particular reason that may now have been lost to time and we are likely to rediscover that reason if we revert it. This is simply the nature of any sufficiently large project as developers come and go and knowledge fades.

Given the decisions on what particular items/issues are worked on aren't ours to make, (and that your request is for an older model of machine and likely to be low priority) my suspicion (and suggestion) is that you might get more expedient results by opening a pull request with your proposed changes to get things working for your needs for potential inclusion in a future firmware.  Then the powers that be can decide to include it during regular integration testing and ensure it doesn't have any undesired side effects.

galagithub, 2022-02-10:
So a bug report comes across as rude, oh my!

If you look I've already provided a solution, so I am a contributor myself and not a Prusa employee.
Secondly you can't tell which are Prusa employees and which are not.

The solution is quite trivial, the code has been in Marlin for a long time where they had subtracted 15 degrees for all MAX_TEMP, again, just because. It was present in 1.1.9 Marlin.

But, in Marlin they just increased the printer MAX_TEMP by 15 in the configuration file and they are not using a LCD menu for PID tune, thus nobody cared about consistency because it was not visible.

Also, it works using gcode, I am printing right now with bed set at 120 degrees.

closed 2023-11-15 by github-actions[bot]