prusa3d/Prusa-Firmware #3387: [BUG] UI - Colour change timer persists after canceled print containing filament change

Stego72, 2022-02-11:
**Printer type** - MK3S
**Printer firmware version** -  3.10.1

**MMU upgrade** - N/A
**MMU upgrade firmware version** - N/A

**SD card or USB/Octoprint** - SD card

**Describe the bug**
If a print containing a filament change is canceled before the change operation  and a new print (without filament change) is started - the main info window still shows the "old" timer for the change operation (eg. 0:03C)
  
**To Reproduce**
1. Start a print containing a filament change
2. After print start (1 layer or more) - cancel print 
3. Start a new print not containing a filament change
  
**Expected behavior**
After starting the second print the remaining time for the filament change for the first print is still shown. No filament change command is executed (i.e. a pure UI issue)

  
**G-code**
  Please attach a G-code. This will make it easier for us to replicate the error.

**Video**
  Please attach a video. It usually helps to solve the problem.

leptun, 2022-02-12:
I'll look into this tomorrow

leptun, 2022-02-13:
I've found the issue and submitted a PR (#3397).
Can you test this as well to see if it's really fixed? Here are the changes on top of 3.10.1 release:
[FW3101-Build4697-fix-time-remaining.zip](https://github.com/prusa3d/Prusa-Firmware/files/8055175/FW3101-Build4697-fix-time-remaining.zip)

PR #3397 (2022-02-15): Fix time remaining not reset on stop print
This broke when we stopped using M84 at the end of a print from the UI (eg: cancel after crash detect, stop print, etc...). So I moved the time remaining reset to the correct function.
Fixes #3387
closed 2022-02-15 by DRracer