prusa3d/Prusa-Firmware #3394: [QUESTION] When compiling firmware, "Board mem has been already modified or not reset"

devsfan1830, 2022-02-12:
I am compiling my own version of the 3.10.1 firmware for a MK2.5/MK2.5S machine with a custom Z height. I've done this for every release using the Windows Linux Subsystem and the build script. I have encountered a message/choice I've not seen before:

Board mem has been already modified or not reset
Current: 0x21FF
1) Yes
2) No
Select Yes if you want to reset it.

What is the safer choice for this and what exactly does this do?

leptun, 2022-02-12:
@devsfan1830 The choice basically anyone should choose is "yes". The setting was added to the script by my request. Because we have a MK3 emulator, we are able to compile the firmware without the flash and memory restrictions in mind. That means we can compile the firmware for imaginary hardware such as an atmega2560 with 32MB of flash and 64K of ram.
This setting is not really well explained unfortunately. You are not the first person to complain about this (@mkbel complained as well). I think the script gives you this prompt if it doesn't have a previous selected value cached in the build env. If this is the first time running the script after pulling changes from upstream, it makes sense that the value was not set previously as it was just introduced.

@3d-gussner Let's discuss and improve these prompts in PF-build.sh

devsfan1830, 2022-02-12:
Yeah, this was a first time run. Thanks for responding so quick!

leptun, 2022-02-19:
I'll keep this issue open until the messages are fixed. I also tagged it for 3.11

3d-gussner, 2022-03-04:
It is fixed in https://github.com/prusa3d/Prusa-Firmware/pull/3422, but we will not add this for FW3.11.0
@devsfan1830 I hope that is okay for you to wait until FW 3.12.0, if so please consider to close this issue.

devsfan1830, 2022-03-04:
Fine by me.

closed 2022-03-04 by devsfan1830