prusa3d/Prusa-Firmware #3401: [Feature Request / Idea] Implement M906 with `tmc2130_val2cur()`

gudnimg, 2022-02-13:
**Is your feature request related to a problem? Please describe.**

This is not a feature I request specifically, just an idea for using unused code. 

When I was looking at the M905 code I noticed that Prusa firmware does not implement M906. But we may already have code tthat can be used to implement M906. See `tmc2130_val2cur()`. https://github.com/prusa3d/Prusa-Firmware/blob/e1ebc826544c9848c3926cba35a605af91cb16d5/Firmware/tmc2130.cpp#L1130

[image]

I have no plans on implementing this myself, please consider this ticket as a *note*. Anyone in the community is of course welcome to try to implement this 😄

leptun, 2022-02-13:
We have M907 that does exactly this (without the I argument).
Btw, it's tmc2130_cur2val, not tmc2130_val2cur. The second one is indeed unused.

gudnimg, 2022-02-13:
@leptun my mistake, I meant M907, M905 isn't available.

leptun, 2022-02-13:
Well, M907 and M906 can be identrical as far as I am concerned. They do basically the exact same thing with TMC drivers. I'd personally leave these as they are now. We can look at this later if we still have space.

closed 2023-06-22 by gudnimg