prusa3d/Prusa-Firmware #3418: [FEATURE REQUEST]  New Gcode for enabling and disabling Crash Detect on the fly.

ggppjj, 2022-02-17:
Please, before you create a new feature request, please make sure you searched in open and closed issues and couldn't find anything that matches.

If it makes sense, enter what type of printer or upgrade the feature request applies to.
**Printer type** - [e.g. MK3S, MK3, MK2.5S, MK2.5, MK2S, MK2]
MK3S+
**MMU Upgrade** - [e.g. MMU2S, MMU2, MMU1]
MMU2S

**Is your feature request related to a problem? Please describe.**
  My printer had a crash detect event while on a lower layer due to no sparse layers on an MMU print, which caused a crash into the printed object.
  
**Describe the solution you'd like**
  A Gcode that allows a slicer to set crash detection would allow slicers to temporarily disable crash detection when objects that are printed individually are being printed at a lower layer height than objects in the path of a re-home.

taikolito, 2022-03-15:
Hi, someone suggested a good-looking solution here, not tested yet: https://forum.prusaprinters.org/forum/original-prusa-i3-mk3s-mk3-hardware-firmware-and-software-help/enable-disable-crash-detection-with-g-code/#post-210996

regards

gudnimg, 2023-07-27:
Closing this issue as resolved by https://github.com/prusa3d/Prusa-Firmware/pull/4053

ggppjj, 2023-07-27:
Excellent, thank you so much!