prusa3d/Prusa-Firmware #3419: [FEATURE REQUEST] New Gcode to set the minimum safe Z height for crash detect rehome.

ggppjj, 2022-02-18:
Please, before you create a new feature request, please make sure you searched in open and closed issues and couldn't find anything that matches.

If it makes sense, enter what type of printer or upgrade the feature request applies to.
**Printer type** - MK3S+
**MMU Upgrade** - MMU2

**Is your feature request related to a problem? Please describe.**
  A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]
  I was printing an MMU print with no sparse layers enabled, and had a crash detect on the wipe tower after it was lower than the printed object in the bottom left of the build plate. It entered a crash detect loop after incorrectly rehoming against the print.

**Describe the solution you'd like**
  A clear and concise description of what you want to happen.
A Gcode that allows the slicer to inform the printer of the minimum safe z-height for rehome would allow the printer to rehome correctly when objects are printed individually.

mmalecki, 2022-07-13:
I've also ran into this problem. Another approach I've pondered is recording maximum Z-height visited during print, and then moving to that + MMU pause Z-lift, but there doesn't appear to be a clean way to approach that.

mmalecki, 2022-07-13:
I hacked [this together](https://github.com/mmalecki/Prusa-Firmware/tree/g60-g61). Will test it over the coming days - initial test was a failure (print didn't resume after MMU OK), but I think the idea may be sound.

Prusa-Support, 2022-07-21:
Hello! Thank you @ggppjj for the request and @mmalecki for sharing the possible solution for this. I brought this to our devs attention, so that they can further review and evaluate this. We'll update here in case of more info from our side - of course, any more feedback anyone might have is absolutely welcome and appreciated! 

Alessandro Pantaleo
Prusa Research

ggppjj, 2022-10-25:
Thank you Alessandro!

It is my hope that this can be evaluated for inclusion on the still-forming 3.13.0 milestone, I am of the opinion that it would be a nice polish to an otherwise excellent feature that would fit nicely into the overall theme of MMU fixes that 3.13.0 appears to target, and also have the added benefit of shoring up the "print objects individually" functionality and could more reliably allow print farm operators to keep automatic crash detection enabled for example (although I admit that the last point is only a hypothetical on my part, not operating or working with anyone operating a print farm myself or knowing their settings preferences).

3d-gussner, 2023-11-07:
@gudnimg Please have a look at this feature request and the approach by @mmalecki https://github.com/prusa3d/Prusa-Firmware/issues/3419#issuecomment-1183326609

3d-gussner, 2024-05-03:
@ggppjj Thanks for the feature request.
In FW 3.14.0 you will be able to use `M125 Z<lift value>` to change the Z lift height.

PR #4679 (2024-05-03): Replace hard coded crash z lift value with Pause Z lift value.
With `M125 Z<value>` you can set the Z lift value via gcode as requested. Fixes #3419
commit 08d26fde (2024-05-03): Replace hard coded crash z lift value with Z lift value.
With `M125 Z<value>` you can set the Z lift value via gcode as requested.
Fixes #3419
closed 2024-05-03 by 3d-gussner