prusa3d/Prusa-Firmware #3421: [ENHANCEMENT] Add 'Yes' and 'No' choices to message: "G-code sliced for a different printer type. Continue?"

gudnimg, 2022-02-20:
**Printer type** - MK3S
**MMU Upgrade** - MMU2S

**Is your enhancement related to a problem? Please describe.**
If a print is started which was sliced for only the MK3S ( and no MMU) but the actual printer has an MMU, then this message will pop up.

[image]

  
**Describe the solution you'd like**
I would like to see a "Yes" and "No" choices. Right now it feels like I don't have a choice because when I click the knob, the printer just continues with the print.

gudnimg, 2022-02-20:
This seems to work but is kinda wonky user experience. I would not consider this an acceptable solution.
[image]

The `Yes` and `No` options don't appear at the same time as the message. And while the print is stopping after selecting `No` option the LCD stays blank until done. With the MMU it takes a few seconds since the printer will pull the filament all the way back up to the FINDA sensor. It is weird to me to see the LCD blank for too long :)

Side note... after booting up the printer I proceeded to select a file from the SD card to print, the message did not appear this time. The printer proceeded to heat up without any issues. It wasn't until I loaded a filament to the nozzle through the LCD that this message appears (after I selected a file to print). 🤔 

I will not spend more time on this today, may visit it again later.

leptun, 2022-02-20:
The printer identified itself as MK3S until the MMU connects, after which it identifies as MK3S_MMU2S. You started the print so quickly that the MMU did not have time to connect. That's why you didn't get the message in that scenario. Luckily, I've got a fix for this in the new MMU2 implementation, so we can ignore that issue for now.
I was also thinking of doing something similar like you did above, but as you said that approach didn't really work. We will investigate this at a later time. Not a critical bug. Just some mild inconvenience.

gudnimg, 2022-02-26:
This issue can be solved with `lcd_show_fullscreen_message_yes_no_and_wait_P` after this PR is merged: https://github.com/prusa3d/Prusa-Firmware/pull/3090

It was a mistake on my part to use `lcd_show_multiscreen_message_yes_no_and_wait_P`

commit df652124 (2022-12-30): Add yes-no choice to M862 warning

Fixes #3421

Change in memory:
Flash: -98 bytes
SRAM: 0 bytes
closed 2022-12-30 by DRracer