prusa3d/Prusa-Firmware #3444: [FEATURE REQUEST] Set volume of beeps

redfast00, 2022-03-14:
**Printer type** - MK3S

**Is your feature request related to a problem? Please describe.**

The alert beeps of the printer in our hackerspace are extremely loud. It would be nice if there were an option to set this volume lower (about the volume of the sound the printer makes while scrolling through the menu for example). This is already possible on the Prusa Mini (https://help.prusa3d.com/ja/article/sound-feedback-mini_151152), and should also be possible (hardware-wise) on the MK3S, since the menu beeps are more quiet.
  
**Describe the solution you'd like**
An extra option in the menu, where you can select the volume of various sounds the beeper makes.

gudnimg, 2022-03-27:
I agree with you. I haven't looked too much into the code yet but I recently turn on the beeper on my printer. The unload sound is annoying loud especially when I am sitting right next to the printer.

3d-gussner, 2022-03-29:
Nice idea. Here some information I have found:
- the mini has the buzzer connected to an analog pin [PA0_BUZZER](https://github.com/prusa3d/Buddy-board-MINI-PCB/blob/master/rev.1.0.0/BUDDY_v1.0.0.pdf) and one resistor of the STM32 which opens the possibility to modify the output.
- the Einsy [(OC3A/AIN1)PE3](https://raw.githubusercontent.com/ultimachine/Einsy-Rambo/1.2e/board/Project%20Outputs/Schematic%20Prints_Einsy%20Rambo_1.2e.PDF) is connected to P2 pin 1 and then connected to the [LCD screen](https://reprap.org/mediawiki/images/7/70/Controller_final_reprapdiscount.pdf) which drives a NPN to activate the Buzzer

I am not an expert but I think you are very limited how you can modify the "loudness" via a mosfet vs real analog pin.

leptun, 2022-03-29:
You can modify the noise output with PWM, but doing PWM on an active buzzer is a terrible idea. Actually, M300 is not something that should be supported on the MK2/3 printers just because of how terrible it sounds. Active buzzers only sound correctly when they are continuously powered. I may be wrong, but I think the Mini buzzer is passive, so it behaves much like a small speaker (so it can be frequency driven). With passive buzzers you can alter the loudness to a certain extent, but it's not amazing there either (but definitely an improvement). To get REAL frequency and volume control, you'd need a DAC pin somehow connected to the buzzer, something we definitely don't have on the 8 bit microcontrollers (and the 32bit board is also not using that apparently).
If you wanted to do loudness control on an active buzzer where you don't try to insert your own frequency, then high frequency PWM could be used (so it only controls the "voltage" of the signal, not also the frequency of the buzzer), but that is not possible on the miniRambo/EinsyRambo since we are doing the buzzer pwm in software using timer interrupts. The added overhead for beeping would be stupidly high, so not worth it.

redfast00, 2022-06-27:
There seem to be multiple volumes already (browsing through the menus is a lot quieter then the filament change beep, the second filament change beep is louder then the first, ...), so I think it is possible to do this from software with the current hardware. It would be nice if it were possible to set the volume of the loud alert beeps during for example filament changing: I still want to get alerted, but the current beep is way too loud to leave on.

Prusa-Support, 2022-07-14:
Thank you for reaching us.

As highlighted by @leptun, the MK3 hardware may not support sound volume adjustments - I do confirm this is possible on the MINI instead.
I'd recommend keeping the issue open as feature request for now, and maybe our developers and collaborators will come out with a neat trick to allow a sort of volume adjustment one day.

I also recognize that the MK3 is capable of reproducing different sounds, in fact, the sound seems different when browsing through the menu, selecting an item and alert notifications. I have the impression though, that the different sounds are only a consequence of longer or shorter sound input.
Maybe, if adjusting the buzzer power is not a viable option, adjusting the duration of the sound input (milliseconds) could make a difference. I'd leave this up to the developers.

For now, be aware of the fact that the MK3 supports various sound modes which are explained in the [Handbook](https://www.prusa3d.com/downloads/manual/prusa3d_manual_mk3s_en.pdf#_ga=2.9972661.1464357517.1657632636-111641450.1649874801).
● Loud - printer uses acoustic signalization for failure or if confirmation is needed
● Once - this level is similar to Loud, but all beeps are played only once
● Silent - most of the beeps are muted, the user is informed only in case of a serious
error
● Assist - accessibility feature which plays a different sound while scrolling through the
menu and at the end of the menu

Alternatively, for now, consider adding padding or material to dampen the sound between the buzzer and LCD cover.

Michele Moramarco
Prusa Research

closed 2023-11-15 by github-actions[bot]