prusa3d/Prusa-Firmware #3445: filament unload

stellr, 2022-03-16:
I was not at my mk3s+, firmware  3.10.0-4481, when the filament ran out on a spool.  The hot end was clogged when I got back to the printer hours later.  After I got things cleared, I tested the senor and it worked great.  The print job paused, the menu items went to zero.  However, the menu prompted me to push the knob to do the unload.  Since I was not there, the PLA continued to cook/burn.  I'm not sure if/when the heat was disabled, but it wasn't fast enough.  I'd like to see the filament unload without manual intervention or have the heat turn off immediately along with the print job pause.

stellr, 2022-03-17:
I'm told there is a 30 minute timeout on the heat on a end-of-filament detection.  I'm testing if my is working as designed right now.

Prusa-Support, 2022-05-10:
Hello! Sorry for the late response to this thread. I confirm the 30-minute heating timeout after the runout detection. I believe that what caused the clog was the combination of having the fil. in the hotend with this hot for some time, followed by the cooling, likely this caused some thermal expansion on the filament causing issues. 

Filament unload without intervention or cooling immediately in that situation: I marked the issue as a feature request, so that this can be better reviewed. 

Alessandro Pantaleo
Prusa Research

pedjas, 2022-08-05:
I believe I run into the same issue the other day. Printer got out of filament I was not there and when I came I was not able to just replace filament as there was some clog in print head. Hours of print was lost. Preventing this to happen would be great.

I also posted other feature request that may help if such clog happens: https://github.com/prusa3d/Prusa-Firmware/issues/3554

Prusa-Support, 2022-08-10:
Hello @pedjas, thank you for your report. I see the situation, but, typically, however, even in case of a clog, it shouldn't be necessary to move the print head to unclog it - it can be very typically done with the techniques here: https://help.prusa3d.com/article/clogged-nozzle-hotend-mk3s-mk2-5s_2008, or in case the clog cannot be pushed through at a high temperature, even with these manual techniques: https://help.prusa3d.com/article/removing-filament-from-extruder-manually_121491. 

I'll review the other thread and pass on the feature request there to devs for them to evaluate. 

Alessandro Pantaleo
Prusa Research

pedjas, 2022-08-11:
Well, i tried everything and I had to move head in position I can have better access.
That is not main issue. Issue is that I had to print form beginning and I believe it is possible to avoid with some firmware improvements.

closed 2024-01-15 by github-actions[bot]